kevinkovalchik/RawTools #1: Problems installing ThermoFischer.CommonCore.RawFileReader

wolski, 2018-09-28:
Hello, Could you please briefly document where to obtain the ThermoFischer.CommonCore.RawFileReader nuget an how to install it with Visual Studio 2017?
This is the error which I got.

[image]

kevinkovalchik, 2018-09-28:
Hello,
I take it you are wanting to look at the source code in Visual Studio? If so, then directions for requesting access to the RawFileReader package are below. If you just want to try using RawTools, then you can download the compiled program from the [releases page](https://github.com/kevinkovalchik/RawTools/releases).

To request access to RawFileReader you need to email [Jim Shofstahl](jim.Shofstahl@thermofisher.com) at Thermo Fisher and request access to their file share site. There are some more details at [PlanetOrbitrap](http://planetorbitrap.com/rawfilereader#.W6471U9lA0M).

Thanks for bringing this up either way. I will add some directions for how to get the RawFileReader nuget package to the readme.

wolski, 2018-09-29:
Dear Kevin

Thanks for the answer. Yes, I would like to be able to build your project with Visual Studio. I have not much experience with C# development however.
Is this correct that you are still using Visual Studio 2015 or would it work also with VS 2017?

Do I need the nuget from Jim Shofstahl or do maybe the dll's you provide in the release suffice?

wolski, 2018-10-01:
Thanks for the information. Managed to build it using the dll's from the realease.

closed 2018-10-01 by wolski