kevinkovalchik/RawTools #3: KeyNotFoundException

sean-ocall, 2018-11-08:
Hi,

I've been finding RawTools extremely useful, but am getting some errors. I'm getting this error when trying to do QC, it happens on about half of my .raw files. The command I'm using (on Ubuntu) is: 

`mono RawTools.exe qc -d <dir containing .raw files>  -q <output_dir>`

Exception is:
`Unhandled Exception:
System.Collections.Generic.KeyNotFoundException: The given key '80512' was not present in the dictionary.`

Full error attached.

Any ideas?

[rawtools_exception.txt](https://github.com/kevinkovalchik/RawTools/files/2562677/rawtools_exception.txt)

chrishuges, 2018-11-08:
Hi Sean,

This looks like it is related to issues linking scans due to missing meta data, something we were made aware of in this [other issue](https://github.com/kevinkovalchik/RawTools/issues/2#issuecomment-434081362). We are planning to push a [new release](https://github.com/kevinkovalchik/RawTools/blob/master/changelog.md) tonight or tomorrow morning that will contain the fix for this (and a couple of other bugs). We will make a post here when it is up so you will be notified and you can try this new version and see if it fixes your problem. 

Cheers,
Chris

chrishuges, 2018-11-09:
Hi Sean,

The new release is posted [here](https://github.com/kevinkovalchik/RawTools/releases/tag/1.3.2). Can you try processing some of your problematic files with this release?

Thanks,
Chris

sean-ocall, 2018-11-12:
Hi Guys,

I've successfully processed my files with the new release - thanks for the response. 
Cheers, Sean

closed 2018-11-12 by sean-ocall

kevinkovalchik, 2018-11-12:
That's good news! Thanks for letting us know.

Kevin