kevinkovalchik/RawTools #4: Identipy doesn't install on Mac

neiladhikari, 2018-11-09:
1) pip install command doesn't work (empty INSTALL file)
2) pip install identify defaults to python3.6. 
Using pip2.7 install identipy fails to install identipy.
3) Cannot call identipy from RawTools script.

kevinkovalchik, 2018-11-09:
Hello,

Sorry you're having issues installing IdentiPy. It isn't our package, so there's limited support I can offer. One thing you could try is running pip directly from python 2.7. So you would invoke something like this:
`$python2.7 -m pip install identipy.zip` where identipy.zip is the downloaded repository.

If this fails, you should probably open an issue on their bitbucket page: https://bitbucket.org/levitsky/identipy
I believe you don't need to register on Bitbucket to create an issue if you don't want to.

Kevin

kevinkovalchik, 2018-11-09:
Actually, I thought of something else:

Did you use `pip2.7 install` or `pip2.7 install .` from inside the downloaded repository?

If you can send me the installation commands and error messages you got when trying to install, I might be able to tell if it is a problem with IdentiPy which you need to ask them about, of if it is a command line syntax problem.

Kevin

neiladhikari, 2018-11-09:
Hi Kevin,
Thanks for pointing that out, pip2.7 install . did work! I guess I wasn't using the correct syntax. I am now able to get rest of the steps in the workflow going, lets see, hoping for the best.
Neil

kevinkovalchik, 2018-11-09:
Great! I'll close this issue then, but feel free to let us know if you run into any problems.

Kevin

closed 2018-11-09 by kevinkovalchik