kevinkovalchik/RawTools #5: Couldn't run RawTools on Windows 7

fazeliniah, 2018-11-21:
Hi,
I get the following error message when I try to run RawTools. I appreciate any help.
Thanks

C:\RawTools1.3.2>RawTools parse -f ch_23Aug2018_HeLa_Std_1.raw -x

Processing: ch_23Aug2018_HeLa_Std_1.raw

Unhandled Exception: System.TypeInitializationException: The type initializer fo
r 'ThermoFisher.CommonCore.Data.Business.RawFileReaderFactory' threw an exceptio
n. ---> System.IO.FileLoadException: ThermoFisher.CommonCore.RawFileReader.dll i
s not found in C:\Users\Owner\Downloads\RawTools1.3.2 ---> System.IO.FileLoadExc
eption: Could not load file or assembly 'file:///C:\Users\Owner\Downloads\RawToo
ls1.3.2\ThermoFisher.CommonCore.RawFileReader.dll' or one of its dependencies. O
peration is not supported. (Exception from HRESULT: 0x80131515) ---> System.NotS
upportedException: An attempt was made to load an assembly from a network locati
on which would have caused the assembly to be sandboxed in previous versions of
the .NET Framework. This release of the .NET Framework does not enable CAS polic
y by default, so this load may be dangerous. If this load is not intended to san
dbox the assembly, please enable the loadFromRemoteSources switch. See http://go
.microsoft.com/fwlink/?LinkId=155569 for more information.
   --- End of inner exception stack trace ---
   at System.Reflection.RuntimeAssembly._nLoad(AssemblyName fileName, String cod
eBase, Evidence assemblySecurity, RuntimeAssembly locationHint, StackCrawlMark&
stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIntro
spection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadAssemblyName(AssemblyName as
semblyRef, Evidence assemblySecurity, RuntimeAssembly reqAssembly, StackCrawlMar
k& stackMark, IntPtr pPrivHostBinder, Boolean throwOnFileNotFound, Boolean forIn
trospection, Boolean suppressSecurityChecks)
   at System.Reflection.RuntimeAssembly.InternalLoadFrom(String assemblyFile, Ev
idence securityEvidence, Byte[] hashValue, AssemblyHashAlgorithm hashAlgorithm,
Boolean forIntrospection, Boolean suppressSecurityChecks, StackCrawlMark& stackM
ark)
   at System.Reflection.Assembly.LoadFrom(String assemblyFile)
   at ThermoFisher.CommonCore.Data.ObjectFactory`1.Initialize(Boolean throwExcep
tions)
   --- End of inner exception stack trace ---
   at ThermoFisher.CommonCore.Data.ObjectFactory`1.Initialize(Boolean throwExcep
tions)
   at ThermoFisher.CommonCore.Data.Business.RawFileReaderFactory.CreateReader()
   at ThermoFisher.CommonCore.Data.Business.RawFileReaderFactory..cctor()
   --- End of inner exception stack trace ---
   at ThermoFisher.CommonCore.Data.Business.RawFileReaderFactory.ReadFile(String
 fileName)
   at RawTools.Program.DoStuff(ParseOptions opts) in C:\Users\Kevin\Documents\GS
C\Projects\RawTools\RawTools\Program.cs:line 329
   at CommandLine.ParserResultExtensions.WithParsed[T](ParserResult`1 result, Ac
tion`1 action)
   at RawTools.Program.Main(String[] args) in C:\Users\Kevin\Documents\GSC\Proje
cts\RawTools\RawTools\Program.cs:line 63

kevinkovalchik, 2018-11-21:
Hello,

Thanks for bringing up the issue. I think windows might be blocking the file, though I'm not sure. This isn't something we have encountered, but we haven't done much testing on Windows 7.

Try this:
1. Right click on RawTools.exe and go to Properties
2. At the bottom of the window next to `Security`, if it says something about the application being blocked, click on the `Unblock` button.

If that doesn't work, see if there is a similar option in the properties of the .dll files in the directory.

To be sure, is your .NET version 4.6.2 or greater?

Kevin

fazeliniah, 2018-11-21:
Thanks for your quick reply. The ,NET version is 4.6.2
I did unblock all the files in the directory and ran it again. I get a shorter error message:

C:RawTools1.3.2>RawTools.exe parse -f ch_23Aug2018_HeLa_Std_1.raw -x

Processing: ch_23Aug2018_HeLa_Std_1.raw

Unhandled Exception: System.NullReferenceException: Object reference not set to
an instance of an object.
   at ThermoFisher.CommonCore.RawFileReader.RawFileAccessBase.SelectInstrument(D
evice instrumentType, Int32 instrumentIndex)
   at RawTools.Program.DoStuff(ParseOptions opts) in C:\Users\Kevin\Documents\GS
C\Projects\RawTools\RawTools\Program.cs:line 333
   at CommandLine.ParserResultExtensions.WithParsed[T](ParserResult`1 result, Ac
tion`1 action)
   at RawTools.Program.Main(String[] args) in C:\Users\Kevin\Documents\GSC\Proje
cts\RawTools\RawTools\Program.cs:line 63

kevinkovalchik, 2018-11-21:
Hm. Well, that's some progress at least!

I can reproduce the error by giving it the name of a raw file that doesn't exist (which is something I should address as well). Can you try using the full path to the file after -f? It is supposed to work with just the filename, but maybe this is another Windows 7 thing we haven't found...

KunathBJ, 2018-11-21:
Hello,

I've got the same errr on windows10 enterprise. Even with the full path and it doesn't seem to be blocked.
So me I just have the long message, not the short one.

Thank you very much,

Benoit

kevinkovalchik, 2018-11-21:
Hi Benoit,

Try replacing the RawTools.exe.config file with the following file:

[RawTools.exe.txt](https://github.com/kevinkovalchik/RawTools/files/2606109/RawTools.exe.txt)

GitHub won't let me upload something with the extension .config, so I changed it to .txt. Be sure to change it back to .config, so it will be called `RawTools.exe.config`

If it is a security issue, which the error message seems to indicate, I think this will fix it. If it works I will make this a permanent change to the config file in the repository.

Kevin

KunathBJ, 2018-11-22:
Hello,

It did fix it!! Very useful output!!
Thanks a lot!
Benoit

kevinkovalchik, 2018-12-03:
@fazeliniah let me know if your issue wasn't resolved and we can reopen this.

Kevin

closed 2018-12-03 by kevinkovalchik

aulke, 2019-03-05:
[image]
Hi Kevin, 
I am also having an issue running RawTools 1.4.2 version on my Win 10 Enterprise PC. RawTools ran just fine until just now.
When I input in cmd  i.e. C:\Users\aulke\Documents\RawTools-1.4.2>RawTools.exe parse -h, I get the error box in the attached file, as well as the message "Access is denied" in cmd. 

I tried the tips from above: Changing RawTools.exe.config as well as "unblocking" all .dll files. I could not unblock the RawTools.exe file since I could not find an "unblock" tick box. 
As an interesting site: RawTools 1.4.1 works fine. 
So far, I could not find a solution and any help would be appreciated, 
Cheers, 
Anne

kevinkovalchik, 2019-03-05:
Hm. The most obvious cause of this would be that your computer is running a 32-bit OS or has a x86 based processor... but I imagine that is not the case since v1.4.1 works fine.

There is very little difference between 1.4.1 and 1.4.2. I didn't change the target OS architecture or anything... Have you tried any of the compatibility modes in the program properties?

The only significant change in 1.4.2 that might cause the system to be unhappy is that the .NET cultural environment is forced to be "invariant" to prevent certain language settings from using comma decimals instead of dot decimals in the MGF file, as X!Tandem seems to have no idea what to do with them. Is your OS language set to something other than English? Possibly that combined with an over-zealous systems security might be causing the issue, but I really don't know...

aulke, 2019-03-05:
I am running 64-bit operating system with x64-based processor. i7, 16GB ram. I have the .NET framework 4.7.2 (461808).