kevinkovalchik/RawTools #7: System.IndexOutOfRangeException during method data extraction

kevinkovalchik, 2018-11-27:
Received the following issue via email:

> 1. the RawTools.exe does nothing at all  if i am trying to launch the programm.
>
>2. Via the commad line it seems to work until the step of processing the raw file, please the the picture for describing the issue below
 > 
>[image]
 >
>Thanks in advance for your help

chrishuges, 2018-11-28:
Updated comments from the user above. 

>It was actually a DIA raw file measured on a Q-Exactive HF. In the meantime I figured out, with DDA runs it works. So far I can do the analysis w/o search, the search did not worked so far, but possibly, there is a simple issue on my side. 

>With https://rawtoolsqcdv.bcgsc.ca/ I often receive the error "Maximum file size exceed" with the [...] Matrix.txt files (> 30Mb).

>It would be great to have the file names somewhere in the graphical outputs of the web browser application (best via a option - e.g. show data point names). The values of the axis are a bit odd.

>At present the matrix.txt are done each per raw file, having the possibility to aggregate them in the web browser application would be quite helpful for reasons of comparisons.
E.g. having the possibility to do a pdf report  with overlay of different raw files or showing them side by side.

chrishuges, 2018-11-28:
Very useful feedback! We will work on implementing these features. 

DIA is currently unsupported with RawTools, but support is in the works for the near future.

chrishuges, 2018-11-29:
> 2.) I totally understand this issue. Nevertheless, the first thin when I showed this tool to a collague was "Oh I can´t see which sample is what..." I think, what one could do is having a checkbox that the user can decide. The user then can clean up the mess via export the graph as pdf  and then edit the labels via illustrator or inkscape or...Or having the files listed in the web interface and the ones the user would like to have labeled he can simply select. The option label outliers is of course also a great idea.

> 3.) Sorry, i wanted to say that it would be cool that the graph trys to use integer value whenever possible. Having the number 245 here makes it more easy to read. A very minor thing of course.

> 4.) Didn´t wanted to say that one should horizontally combine them. What I wanted to say is, that it would be cool to have the graphs derived from different matrix.txt in one presentation.
Consider you want to compare different runs via that method, At present I have to make a graph separatedly for every raw file, export it and combine the graphs manually.

chrishuges, 2018-11-29:
More useful feedback. We have implemented some features in the R Shiny app to address these concerns, including greater plot customization and the ability to make 'multiple plots' in the same space. This updated version of the shiny app is already on the GitHub and will appear on the rawtools website soon.

chrishuges, 2018-11-30:
User pointed out some errors in the Shiny app that have now been fixed and uploaded to the web.

chrishuges, 2018-12-04:
User suggested changing some of the plots to display density data. Have updated these plots in the Shiny application (on GitHub). The updated version will go live on the web shortly.

chrishuges, 2018-12-06:
Per the user suggestion, we have added overlaid chromatograms to the Shiny application.

closed 2018-12-08 by chrishuges