kevinkovalchik/RawTools #8: Add link to installation on the main page

SamGG, 2018-11-28:
Hi,
Thanks for your article and these interesting tools.
Could you add an installation paragraph at the top of the main README (https://github.com/kevinkovalchik/RawTools)? I wanted to install RawTools quickly and the installation procedure was not straightforward to me.
Best.

chrishuges, 2018-11-28:
Hi,

Thanks for the suggestion. The steps for getting RawTools running on your system are going to depend on the operating system you are using. We have detailed these steps on our wiki page [here](https://github.com/kevinkovalchik/RawTools/wiki). 

Per your suggestion, we have added some text to the main page under the 'RawTools releases and installation' heading directing the users to this wiki.

Cheers,
Chris

SamGG, 2018-11-28:
Super. That will help a lot.
BTW, you should decrease the RAM requirement for Linux ;-)

chrishuges, 2018-11-28:
What do you mean regarding the RAM requirement?

SamGG, 2018-11-28:
I think there is a **one** typo: **1**32 GB

chrishuges, 2018-11-28:
Hi,

That actually isn't a typo. The linux system we were using in that walkthrough has 132GB of RAM. However, that is not a 'requirement' for RawTools, that is just the system we used for the walkthrough. The system requirements are detailed a bit higher up in the wiki. I have changed the walkthrough text to address any potential confusion here.

Thanks for noticing this!