kevinkovalchik/RawTools #9: Installation fails: Error in htmlDependency("font-awesome", "5.3.1", "www/shared/fontawesome",  :

SamGG, 2018-11-28:
Starting from a fresh R 3.5.1 on Windows 7 x64 with Rtools35, we installed the various packages needed by RawTools
```
install.packages(c("shiny", "ggplot2", "reshape2", "RColorBrewer", "data.table", "shinydashboard"))
```
When running the shiny app, an error appears:
```
Error in htmlDependency("font-awesome", "5.3.1", "www/shared/fontawesome",  : 
unused argument (package = "shiny")
```
[image]

This error has been referenced at https://community.rstudio.com/t/r-shinydashboard-instructions-doesnt-work/18000/3. Installing the latest htmltools v 0.3.6 from source solves the error.
```
install.packages("htmltools")
```
For installation from source, you must install Rtools with administrative permissions and add R tools to the PATH.
HTH

chrishuges, 2018-11-28:
Thanks for the heads up. I just added some code to the Shiny app to check whether all required libraries are present, and if not, to install them (including htmltools). Will post the updated version here in the next day or so.

Note for RStudio users - admin is not required to run this package through RStudio. 

Cheers,
Chris

SamGG, 2018-11-28:
Thanks. I think you should let the issue open until your really updated the repo in order to allow users to catch the solution here.
Best.

chrishuges, 2018-11-28:
Sorry, when I saw the updated version will be posted in the next day or so, I meant in the online web interface. The source code version is posted. I should have been more specific.

chrishuges, 2018-11-28:
True, good catch! I will fix this and update the source file. It should always install the latest htmltools, so I will skip the versioning for now unless it becomes an issue for someone in the future.

closed 2018-12-08 by chrishuges