kevinkovalchik/RawTools #10: failed quality control with search

KunathBJ, 2018-11-29:
Hello,

I've been trying to run the QC with the search but I get several errors and I can't figure out if they're all linked or different. The first one being that the software can't find cython but I gave the path to pyhton and to the identipy-script. Shall I also give the part for cython?
I've pasted the line I used as well as the error. I'm working on windows10 and can run the QC without search without any issues.

Thanks a lot for your help!
Ben

`C:\Users\benoit.kunath\Documents\Software\RawTools1.3.2>RawTools.exe qc -d MP_Symp\ -q MP_Symp\qc_output2 -s identipy --db MP_Symp\ovgu_prodigal_noEnds_human.faa -p C:\Python27\python.exe -I C:\Python27\Scripts\identipy-script.py -N 10000
Checking Identipy dependencies
All Identipy dependencies satisfied!
5 file(s) to process
Processing MP_Symp\OVGU_Rep1.raw
Indexing linked scan events: 100%
Extracting raw data: 100%
Formatting scan meta data: 100%
Analyzing precursor peaks: 100%
Calculating peak symmetries: 100%
Writing MGF file: 100%
Starting Identipy
   INFO: [14:12:17] Cython modules were not loaded...
   INFO: [14:12:17] Reading defaults from C:\Python27\lib\site-packages\identipy\default.cfg
   INFO: [14:12:17] Reading config from C:\Users\benoit.kunath\Documents\Software\RawTools1.3.2\orbitrap_config.cfg
   INFO: [14:12:48] Setting legend: {}
Traceback (most recent call last):
  File "C:\Python27\Scripts\identipy-script.py", line 11, in <module>
    load_entry_point('identipy==0.2', 'console_scripts', 'identipy')()
  File "C:\Python27\lib\site-packages\identipy\cli.py", line 242, in run
    utils.write_output(inputfile, settings, main.process_file(inputfile, settings))
  File "C:\Python27\lib\site-packages\identipy\utils.py", line 1461, in write_output
    return writer(inputfile, settings, results)
  File "C:\Python27\lib\site-packages\identipy\utils.py", line 1058, in write_pepxml
    with open(filename, 'w') as output:
IOError: [Errno 2] No such file or directory: 'MP_Symp\\qc_output2\\QcSearchData\\OVGU_Rep1.raw.pep.xml'

Unhandled Exception: System.IO.DirectoryNotFoundException: Could not find a part of the path 'C:\Users\benoit.kunath\Documents\Software\RawTools1.3.2\MP_Symp\qc_output2\QcSearchData\OVGU_Rep1.raw.pep.xml'.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.FileStream.Init(String path, FileMode mode, FileAccess access, Int32 rights, Boolean useRights, FileShare share, Int32 bufferSize, FileOptions options, SECURITY_ATTRIBUTES secAttrs, String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize)
   at System.Xml.XmlDownloadManager.GetStream(Uri uri, ICredentials credentials, IWebProxy proxy, RequestCachePolicy cachePolicy)
   at System.Xml.XmlUrlResolver.GetEntity(Uri absoluteUri, String role, Type ofObjectToReturn)
   at System.Xml.XmlTextReaderImpl.FinishInitUriString()
   at System.Xml.XmlReaderSettings.CreateReader(String inputUri, XmlParserContext inputContext)
   at System.Xml.Linq.XElement.Load(String uri, LoadOptions options)
   at RawTools.QC.SearchQC.LoadSearchResults(QcParameters qcParameters, RawDataCollection rawData) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\QC\SearchQC.cs:line 58
   at RawTools.QC.QC.DoQc(QcParameters qcParameters) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\QC\QC.cs:line 245
   at RawTools.Program.DoStuff(QcOptions opts) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\Program.cs:line 219
   at CommandLine.ParserResultExtensions.WithParsed[T](ParserResult`1 result, Action`1 action)
   at RawTools.Program.Main(String[] args) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\Program.cs:line 63`

kevinkovalchik, 2018-11-29:
Hi again,

I tracked down the problem. The paths to the qc directory (-q) and FASTA file (--db) need to be absolute, not relative. Python and RawTools will be running from different locations, so the relative path doesn't mean the same thing to both of them. I'll look into changing this so relative paths are okay, but for now use absolute paths.

Also, if you have the installation of python 2.7 which your are using in your system path (i.e. you can type `python` in the command prompt from anywhere and python 2.7 loads up) then you shouldn't need to specify the location of python and identipy. RawTools will look for them. Unless, of course, you already tried that and it failed!

In case you didn't know, RawTools also works with X! Tandem. The choice it up to you, of course. However, the interface to run the X! Tandem search is more straightforward.

Hope that helps!

kevinkovalchik, 2018-11-29:
I'll update the command line help to indicate these arguments need to be absolute paths.

KunathBJ, 2018-11-30:
Hello!

Thanks a lot. I'll try that right away.
I specified the location of python and Identipy in an attempt to fix the issue.
I will try with absolute path and and without specifying python (it's in my system path).

Thanks a lot! I'll confirm later if it works for me!

kevinkovalchik, 2018-12-10:
Hi Ben,

How did it work out? Are things okay now or should I revisit the issue?

Thanks,
Kevin

KunathBJ, 2018-12-10:
Hey

Very sorry for the late answer. I was out of office and completely forgot.
So I tried with full path but I have the same error:

`C:\Users\benoit\Documents\Software\RawTools1.3.2>RawTools.exe qc -d MP_Symp\ -q C:\Users\benoit\Documents\Software\MP_Symp\qc_output2 -s identipy --db C:\Users\benoit\Documents\Software\MP_Symp\ovgu_prodigal_noEnds_human.faa -N 10000
Checking Identipy dependencies
All Identipy dependencies satisfied!
5 file(s) to process
Processing MP_Symp\OVGU_Rep1.raw
Indexing linked scan events: 100%
Extracting raw data: 100%
Formatting scan meta data: 100%
Analyzing precursor peaks: 100%
Calculating peak symmetries: 100%
Writing MGF file: 100%
Starting Identipy
   INFO: [00:13:39] Cython modules were not loaded...
   INFO: [00:13:39] Reading defaults from C:\Python27\lib\site-packages\identipy\default.cfg
   INFO: [00:13:39] Reading config from C:\Users\benoit\Documents\Software\RawTools1.3.2\orbitrap_config.cfg
Traceback (most recent call last):
  File "C:\Python27\Scripts\identipy-script.py", line 11, in <module>
    load_entry_point('identipy==0.2', 'console_scripts', 'identipy')()
  File "C:\Python27\lib\site-packages\identipy\cli.py", line 242, in run
    utils.write_output(inputfile, settings, main.process_file(inputfile, settings))
  File "C:\Python27\lib\site-packages\identipy\main.py", line 34, in process_file
    if add_decoy and utils.is_db_target_only(settings):
  File "C:\Python27\lib\site-packages\identipy\utils.py", line 946, in is_db_target_only
    for prot in fasta.read(db):
  File "C:\Python27\lib\site-packages\pyteomics\auxiliary\file_helpers.py", line 182, in helper
    return FileReader(args[0], _mode, _func, True, args[1:], kwargs, kwargs.pop('encoding', None))
  File "C:\Python27\lib\site-packages\pyteomics\auxiliary\file_helpers.py", line 144, in __init__
    self.reset()
  File "C:\Python27\lib\site-packages\pyteomics\auxiliary\file_helpers.py", line 149, in reset
    self._source = _file_obj(self._source_init, self._mode, self._encoding)
  File "C:\Python27\lib\site-packages\pyteomics\auxiliary\file_helpers.py", line 70, in __init__
    self.file = codecs.open(f, mode, encoding)
  File "C:\Python27\lib\codecs.py", line 898, in open
    file = __builtin__.open(filename, mode, buffering)
IOError: [Errno 2] No such file or directory: 'C:\\Users\\benoit\\Documents\\Software\\MP_Symp\\ovgu_prodigal_noEnds_human.faa'

Unhandled Exception: System.IO.FileNotFoundException: Could not find file 'C:\Users\benoit\Documents\Software\MP_Symp\qc_output2\QcSearchData\OVGU_Rep1.raw.pep.xml'.
   at System.IO.__Error.WinIOError(Int32 errorCode, String maybeFullPath)
   at System.IO.FileStream.Init(String path, FileMode mode, FileAccess access, Int32 rights, Boolean useRights, FileShare share, Int32 bufferSize, FileOptions options, SECURITY_ATTRIBUTES secAttrs, String msgPath, Boolean bFromProxy, Boolean useLongPath, Boolean checkHost)
   at System.IO.FileStream..ctor(String path, FileMode mode, FileAccess access, FileShare share, Int32 bufferSize)
   at System.Xml.XmlDownloadManager.GetStream(Uri uri, ICredentials credentials, IWebProxy proxy, RequestCachePolicy cachePolicy)
   at System.Xml.XmlUrlResolver.GetEntity(Uri absoluteUri, String role, Type ofObjectToReturn)
   at System.Xml.XmlTextReaderImpl.FinishInitUriString()
   at System.Xml.XmlReaderSettings.CreateReader(String inputUri, XmlParserContext inputContext)
   at System.Xml.Linq.XElement.Load(String uri, LoadOptions options)
   at RawTools.QC.SearchQC.LoadSearchResults(QcParameters qcParameters, RawDataCollection rawData) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\QC\SearchQC.cs:line 58
   at RawTools.QC.QC.DoQc(QcParameters qcParameters) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\QC\QC.cs:line 245
   at RawTools.Program.DoStuff(QcOptions opts) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\Program.cs:line 219
   at CommandLine.ParserResultExtensions.WithParsed[T](ParserResult`1 result, Action`1 action)
   at RawTools.Program.Main(String[] args) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\Program.cs:line 63`

I find strange that it reports that:
`IOError: [Errno 2] No such file or directory: 'C:\\Users\\benoit\\Documents\\Software\\MP_Symp\\ovgu_prodigal_noEnds_human.faa`
with the double backslash... I didn't write that. Maybe I'm making an error in the way I give the full path??

Thanks
Ben

kevinkovalchik, 2018-12-11:
Hi Ben,

No worries about the delay!

I don't think the double backslashes are an issue. I'm not sure why they would be showing up in only that single line of the error, but double backslashes and forward slashes are both used in place of backslashes in paths in a lot of coding, so I think that isn't the problem. But I think I did track the problem down.

I almost never use relative paths when running RawTools, and a consequence is I almost never found the bugs associated with them! But I think I've got this one working now. I was able to recreate your errors and it was indeed a problem RawTools was having with the paths. The file below should work with relative paths for everything (i.e. -d, -q, --db can all be relative):

[RawTools_IdentipyPaths_20181210.zip](https://github.com/kevinkovalchik/RawTools/files/2665468/RawTools_IdentipyPaths_20181210.zip)

If it works, let me know and I'll include it in the next release (note the fix is built on top of [1.4.0-beta](https://github.com/kevinkovalchik/RawTools/releases/tag/1.4.0-beta), so there will be some differences to the console output and what's happening in the background).

I'd like to check something... I noticed it looks like you are pointing the program to a fasta file in slightly difference locations in the two outputs you sent:

1. C:\Users\benoit.kunath\Documents\Software\RawTools1.3.2\MP_Symp\ovgu_prodigal_noEnds_human.faa
2. C:\Users\benoit\Documents\Software\MP_Symp\ovgu_prodigal_noEnds_human.faa

In the first, the `MP_Symp` folder is located inside `Software\RawTools1.3.2`, while in the second it is just in `Software`. I don't know if pointing it to the wrong place for the fasta file would give the same sort of error, but I just wanted to make sure that wasn't part of the problem!

Kevin

KunathBJ, 2018-12-11:
Hey Kevin,

It is now way better I think. I didn't have to put any full path and it seems to work now.
Yet the cython modules were not loaded but the analysis kept on going.
I didn't change anything regarding the different software in my path for this new version. See attached the code and a screenshot of my paths
[image]

```
C:\Users\benoit.kunath\Documents\Software\RawTools_IdentipyPaths_20181210>RawTools.exe qc -d MP_Symp\ -q MP_Symp\qc_output2 -s identipy --db MP_Symp\ovgu_prodigal_noEnds_human.faa -N 10000
Checking Identipy dependencies
All Identipy dependencies satisfied!
5 file(s) to process

Processing C:\Users\benoit.kunath\Documents\Software\RawTools_IdentipyPaths_20181210\MP_Symp\OVGU_Rep1.raw
Determing MS analysis order... Done!
Extracting scan indices: 100%
Checking for orphaned scans: 100%

================ Scan indexing report ================
Total scans in file: 84591
Scans linked: 84591

Orphan scans:
None!

All scans accounted for!
======================================================

Extracting reaction events: 100%
Extracting scan data: 100%
Extracting trailer extras: 100%
Extracting precursor masses: 100%
Extracting retention times: 100%
Analyzing precursor peaks: 100%
Calculating meta data
  MS1 isolation interference
  MS2 scan cycle density
  Ion injection time
  Duty cycle
  Intensity distribution
  Summed intensities
Calculating metrics
Writing MGF file: 100%
Starting Identipy
   INFO: [10:17:41] Cython modules were not loaded...
   INFO: [10:17:41] Reading defaults from C:\Python27\lib\site-packages\identipy\default.cfg
   INFO: [10:17:41] Reading config from C:\Users\benoit.kunath\Documents\Software\RawTools_IdentipyPaths_20181210\orbitrap_config.cfg
   INFO: [10:17:53] Setting legend: {}
   INFO: [10:17:53] Writing C:\Users\benoit.kunath\Documents\Software\RawTools_IdentipyPaths_20181210\MP_Symp\qc_output2\QcSearchData\OVGU_Rep1.raw.pep.xml ...
   INFO: [10:17:53] Reading spectra ...
   INFO: [10:18:14] 65135 spectra pass quality criteria.
   INFO: [10:18:14] Setting legend: {}
   INFO: [10:18:14] Running the search ...
```

kevinkovalchik, 2018-12-11:
Hi Ben,

Glad it is working! Thanks for bringing up this issue in the first place. Having the relative paths working is a big improvement.

As for the Cython modules not loading, there's not much support I myself can provide. RawTools is just starting IdentiPy from the command line, and getting the Cython modules running is not part of that call. I don't know much about what IdentiPy does behind the scenes. I would suggest you open an issue on the [IdentiPy bitbucket page](https://bitbucket.org/levitsky/identipy). If you need any information from me in regards to how RawTools is calling identipy, let me know.

I notice that you set -N to 10,000, but from the number of spectra that pass quality criteria in IdentiPy it looks like all the spectra are being searched. I'm going to open another issue with that in it, if you are interested.

Take care,
Kevin

closed 2018-12-11 by kevinkovalchik