kevinkovalchik/RawTools #14: Critical: Most of the parent ion info failed to be exported from Q Exactive Plus raw file

microdou, 2018-12-06:
**Describe the bug**
Most of the parent ion info (`MonoisotopicMass`, `ParentPeakArea`, `PeakFirstScan`, `PeakMaxScan`, `PeakLastScan`, `BaseLinePeakWidth(s)`, `PeakParentScanIntensity`, and `PeakMaxIntensity`) is missing when trying to export the `_parse.txt` (or `_Matrix.txt` the old version) file from raw file.

Also note that `ParentPeakFound` field is largely identified as `False`.

**Description of raw file**
MS2

**Command line arguments**
`mono tools/RawTools/RawTools.exe parse -f data/example.raw -p`

**Raw Files**
I've share the raw and exported files in Dropbox:
https://www.dropbox.com/sh/x4dvsos376qjegj/AAAB0ZSdGBI8mraUcLRKZ_N7a?dl=0
The missing parent ion info can be easily identified in `example.raw._parse.txt` or `example.raw._parse.csv`

**Desktop (please complete the following information):**
 - OS: Arch Linux

**Additional context**
This bug occurs when trying to export raw files of `Q Exactive Plus`, while `Fusion Lumos` raw files work fine. So I suppose it's an instrument specific bug.

kevinkovalchik, 2018-12-06:
Thanks for bringing up the issue. I'll have a look at the raw file this
evening and see if I can get it fixed.

By chance, is monoisotopic precursor selection turned off in the QE
instrument method? This causes the monoisotopic mass in the scan header to
be 0 or incorrect, so RawTools can't find the peaks. If so you'll need to
use the -R flag to refine the precursor mass. If this is the issue, it
might be fixed in the next beta release. I've decided to identify precursor
peaks using the trigger mass rather than the monoisotopic mass as it is
generally higher in intensity.

Kevin

microdou, 2018-12-06:
@kevinkovalchik Thanks for the prompt response!

1. I'm not aware of the `onoisotopic precursor selection` switch. What I can see from the method that might be relevant are:
```
Full MS: Spectrum Data Type: Centroid
dd-MS2: Spectrum Data Type: Centroid
dd Settings: Exclude isotope: on
```
2. As you can see, there are still several MS2 scans whose parent ions are detected by RawTools, so it's likely the "killswitch" was not turned on 😂

3. Good to know there is an `-R` parameter, I will look into it.

4. Yes, the trigger mass or most abundant monoisotopic mass with higher intensity may be more relevant in some quality control.

Thanks again!

microdou, 2018-12-06:
@kevinkovalchik Tried the `-R` parameter, it didn't save the parent ion info... You may try playing with my raw file. Thanks!

microdou, 2018-12-06:
@kevinkovalchik By the way, it may be better to export both the `trigger mass`- and `monoisotopic mass`- related info to a txt file, and let users to decide which to use, or just provide the options as parameters.

microdou, 2018-12-06:
@chrishuges Oh yes, found it! Thank you!
`dd Settings: Peptide Match: -`

Its options are: `-`, `on`, or `preferred`. I suppose `-` may be defaulted as `off`...

chrishuges, 2018-12-06:
Yes the default is off. 'On' is the same as having MIPS enabled. 'Preferred' is a hybrid.

kevinkovalchik, 2018-12-07:
1.4.0-beta.2 has been released, and should address this issue. The precursor peaks are now extracted based on the picked mass. I tested out your example file and it seems to go fine now! I agree that using the most intense peak in the isotope cluster would be nice, so this is something I will look into adding as an option.

closed 2018-12-07 by kevinkovalchik