kevinkovalchik/RawTools #17: MGF compatibility with MS-Amanda

verheytb, 2018-12-20:
Hi,

I've been using MGF files generated from RawTools successfully with several search engines. However, with MS Amanda, I get the following error:

```
Searching file 'filename.raw.mgf'...
Error parsing mgf file at or after spectrum 'Spectrum_3'.
Input string was not in a correct format.
Search failed.
```

Parsing of the MGF files is likely the issue. When I look at the MGF file specifications [here](http://www.matrixscience.com/help/data_file_help.html) it looks like the issue is the CHARGE line, which should be `CHARGE=2+` instead of `CHARGE=2`.

kevinkovalchik, 2018-12-20:
Hello,

Thanks for bringing this to our attention. We've tried to follow the specifications you mention, but since a lot of search engines try to correct any errors I sometimes don't know if something is wrong! I'll get this fixed soon.

Thanks,
Kevin

verheytb, 2018-12-20:
I appreciate it, thanks!

Ted

kevinkovalchik, 2018-12-20:
Hi Ted,

Try this out:

[RawTools_FixMgf_20181220.zip](https://github.com/kevinkovalchik/RawTools/files/2700854/RawTools_FixMgf_20181220.zip)

Its based on the beta.4 version, so if you are using an earlier version it will look a little different on the console. There are some detail [here](https://github.com/kevinkovalchik/RawTools/releases/tag/1.4.0-beta).

Kevin

verheytb, 2019-01-02:
Hi Kevin, MS-Amanda seems to be working fine now. Thanks!

closed 2019-01-02 by verheytb

kevinkovalchik, 2019-01-02:
Thanks for letting me know!

Kevin