kevinkovalchik/RawTools #18: Does the quan function support MS2 scan acquired in profile mode?

Leo838, 2018-12-21:
Hello,
Thanks a lot for developing such a handy tool!! But it seems that the current version doesn't support raw files with MS2 scans acquired in profile mode. The reporter quantification works nicely with centroid raw files on my computer, but it complains when I input raw files in profile mode:

Determing MS analysis order... Done!
Extracting scan indices: 100%
Checking for orphaned scans: 100%

================ Scan indexing report ================
Total scans in file: 40704
Scans linked: 40704

Orphan scans:
None!

All scans accounted for!
======================================================

Extracting reaction events: 100%
Extracting scan data: 100%
Extracting trailer extras: 100%
Extracting precursor masses: 100%
Extracting retention times: 100%
Analyzing precursor peaks: 100%
Refining precursor charge and monoisotopic mass: 100%
Calculating meta data
  MS1 isolation interference
  MS2 scan cycle density
  Ion injection time
  Duty cycle
  Intensity distribution
  Summed intensities
Quantifying reporter ions: 0%
Unhandled Exception: System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at RawTools.Algorithms.QuantifyReporters.QuantifyOneScan(CentroidStreamData centroidStream, String labelingReagent) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\Algorithms\QuantifyReporters.cs:line 0
   at RawTools.Algorithms.Quantification.Quantify(CentroidStreamCollection centroidScans, SegmentScanCollection segmentScans, WorkflowParameters parameters, MethodDataContainer methodData, ScanIndex index) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\Algorithms\QuantifyAllScans.cs:line
 58
   at RawTools.WorkFlows.WorkFlowsDDA.ParseDDA(IRawFileThreadManager rawFileThreadManager, WorkflowParameters parameters) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\WorkFlows\DDA.cs:line 94
   at RawTools.Program.DoStuff(ParseOptions opts) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\Program.cs:line 469
   at CommandLine.ParserResultExtensions.WithParsed[T](ParserResult`1 result, Action`1 action)
   at RawTools.Program.Main(String[] args) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\Program.cs:line 66
----------------------------------------------------------------------------------------------
Could you please work on that? Thank you very much.

kevinkovalchik, 2018-12-21:
Hello,

Yes, RawTools expects centroid data for the quantification scans (either Ms2 or Ms3). Addressing the case where a user has profile data is on our to-do list, but we haven't had time to do much with it. RawFileReader has built in functionality to centroid a profile scan, but we haven't had much chance to test it out.

Would you be able to share one of your profile raw files with us to help develop this? Our methods are always set to collect centroid data from both mass analyzers, so we don't have too many profile test files handy. I assume your MS2 data is from the ion trap, right?

I'll be on vacation next week, so I won't be able to look at this further until January.

Take care,
Kevin

Leo838, 2018-12-21:
Thanks for your reply. Yes, we can share the raw files with no problem, and no, all scans were acquired in Orbitrap. Do different MS analyzers make any different?
Interestingly, I have tested numerous files and found that the tool works for profile MS3 from Fusion. It even works nicely with a profile MS2 file I got from an HF-X. Why is that?
One more thing, when processing with MS3 files, the tool introduces a column shift in the matrix, please see attached.   
[MS3_Matrix.txt](https://github.com/kevinkovalchik/RawTools/files/2704252/MS3_Matrix.txt)

kevinkovalchik, 2018-12-21:
Yeah, the data gets saved differently for the IT and OT analyzers. The IT
only saves one stream of data, which is masses and intensities. If the IT
is run in profile mode it saves the profile data, and if it is run in
centroiding mode it keeps the centroid data and discards the profile. The
OT usually saves two data streams: mass and intensity lists for the profile
data, and then also centroid data. The OT always saves both (unless
centroiding is turned off, I suppose). That is why I thought you were
acquiring Ms2 in the IT, but I guess you have the centroiding turned off
for the Orbitrap? If so, can I ask why? I am not an applications expert, so
I'm not sure in what situation you would want to do so.

If you could share the file that is not working that would be great!

Thanks for letting me know about the ms3 matrix issue. I must be getting
too lazy about testing everything before I release it! Is this one of the
betas you are running?

Kevin