kevinkovalchik/RawTools #19: Bug in RTINSECONDS

CarlosHorro, 2019-01-09:
Hi,

Testing RawTools 4.0.0 converting a raw file to a mgf (mono RawTools.exe parse -f qExactive01819.raw -m -R), I think I found a little error in the way RTINSECONDS is calculated.

Our raw file has a scan with RTINSECONDS = 1184.46996 , but when the mgf file is generated by RawTools this scan has a RTINSECONDS = 19.741166.
As 1184,46996 / 60 = 19.741166 , it seems that RawTools is converting the retention time to minutes before writing this number to RTINSECONDS. Due to the name of this attribute, it seems that the retention time should be kept in seconds instead of being converted to minutes.

Greetings,
Carlos

kevinkovalchik, 2019-01-10:
Whoops. You're right, we are definitely reporting minutes in the MGF file. Thanks for catching that! I'll push an update in the next couple days fixing this.

Kevin

kevinkovalchik, 2019-01-12:
Okay, this should be fixed now!
The version containing the fix is [1.4.1](https://github.com/kevinkovalchik/RawTools/releases/tag/1.4.1).

Kevin

closed 2019-01-12 by kevinkovalchik