kevinkovalchik/RawTools #20: Galaxy tool for RawTools?

CarlosHorro, 2019-01-10:
Hi,

I'm Carlos Horro, I work with Harald Barsnes on SearchGUI and PeptideShaker (and their Galaxy tools) and we are very interested on using RawTools on our workflows running on Galaxy when receiving RAW files instead of MGF ones from our users. 
We would like to ask you if you have plans to develop a Galaxy wrapper for RawTools. We think it would be great to collaborate in such a tool, which should not be too complex to be implemented as we already implemented some others successfully.

Hoping that you will find it interesting,
greetings,
Carlos

kevinkovalchik, 2019-01-10:
Hi Carlos,

It is not something we have previously considered, but it sounds like a very interesting idea and we would be excited to pursue this.

Kevin

kevinkovalchik, 2019-01-16:
Hi Carlos,

I am not familiar with Galaxy beyond having heard about what it does. What would be the first steps to develop a wrapper for RawTools? Do we just need to define the tool in an XML file (as is https://planemo.readthedocs.io/en/latest/writing_standalone.html) or is there more to it?

Kevin

CarlosHorro, 2019-01-16:
Hi Kevin,

Well, the entire workflow is a bit complex but it's just about going step by step. I attach a graph I did as a guide for deploying SearchGUI & PeptideShaker Galaxy tools some time ago with all the different steps.

https://docs.google.com/drawings/d/1UScPZ-Qfo9h8aPgmr0WZQ91babavGT0E19nB5rX3bcM/edit?usp=sharing 

So, the first step is to create a Conda package in BioConda for making RawTools automatically distributable (of course, this step by itself is useful far beyond Galaxy).
Having the Conda package working, it will the time for developing the Galaxy tool (in the galaxy-p repo) pointing to the Conda package and, yes, creating the xml mentioned in that doc.
Overall, it's a slow process although it does not require much effort; I'm a bit busy right now  but I can still start to manage the Conda package stuff.
Tell me any doubt you may have about this all please.

Greetings,
Carlos

kevinkovalchik, 2019-01-18:
Hi Carlos,

I'm also pretty busy at the moment. I'll see what I can get done on the Conda package when I get a chance. It'll be nice to know how to do that part so we can update it with new releases. I'll let you know if I have any troubles.

Kevin

CarlosHorro, 2019-01-25:
Hi Kevin,

I have made a PR to the Bioconda repo with the new recipe for RawTools:

#13221

I think there is still something wrong with it as, when I execute a full test, there are some errors I have not been able to solve. But I wanted to have some feedback by the bioconda team before wasting more time looking into it...

Greetings,
Carlos

kevinkovalchik, 2019-01-26:
Hi Carlos,

I realized today that the RawFileReader license might not allow us to
redistribute it this way. I will need to check with Thermo to see if it is
okay.

Kevin

CarlosHorro, 2019-01-28:
Ok. Of course we would include whatever information they require into the Conda package or the Galaxy tool if that makes them to allow these tools.

Greetings,
Carlos

CarlosHorro, 2019-01-28:
By the way, I have found one Conda package which also contains ThermoRawFileReader, so I think it will be possible to use it in this way:
https://anaconda.org/bioconda/thermorawfileparser/files

kevinkovalchik, 2019-01-31:
> @bgruening <https://github.com/bgruening> I think you may already know
> about this as moFF (https://github.com/compomics/moFF/) also uses this
> library?

Hi Carlos,

Thanks for following up on this. I appreciate it. Sorry I haven't gotten to
it myself. I've been busy with thesis writting and fellowship applications,
so haven't had much extra time lately.

Kevin

bgruening, 2019-01-31:
@CarlosHorro let's put it this way, I'm not aware of any licensing issue. But that does not mean much :(

@ypriverol do you know if thermorawfileparser has some restrictions in regards of license and redistribution?

ypriverol, 2019-01-31:
No really. But we haven't look into it.

kevinkovalchik, 2019-01-31:
The license allows the distribution of RawFileReader as part of a package,
but it cannot be redistributed at that point. I'm not sure if us making it
accessible through Conda and then Galaxy is just us distributing it, or if
it constitutes a redistribution in some legal sense.

When people acquire a project through Conda, is it coming straight from the
original source (e.g. release on GitHub) and Conda just makes it easy for
them to manage it?

I'll email Jim Shofstahl at Thermo today and ask about distributing RFR as
part of a package via Conda. It's probably okay...

Kevin

kevinkovalchik, 2019-02-05:
No word yet about the license issue, but I just found the email in my drafts folder, so maybe that isn't too surprising...

Now that I have _actually_ sent the email, I'll let you know when I get a response.

Kevin

CarlosHorro, 2019-02-21:
Any further news about this licensing topic?

kevinkovalchik, 2019-02-21:
Yes, heard back from them last week but haven't had time to follow up on it. We were sent a letter which is basically a consent form from Thermo saying it is okay for us to redistribute RawFileReader. The form is intended for commercial products, so we asked if we actually needed it since our product is not commercial. It wasn't very clear from the response if we actually need the form, but they said it would cover us even though the product is non-commercial.

For the consent form, all they needed from us is a signature from someone who legally represents the organization in such matters. We haven't gotten around to sending it to the legal department yet.

So my interpretation, for anyone else using RawFileReader in this fashion, is that Thermo is not actively looking for people distributing it as part of non-commercial software. They are more worried about people exploiting it for financial gains without telling them first. However, they do have an avenue for officially getting permission to distribute it however you see fit (the EULA only covers p2p distribution, I think) and they make the process pretty easy. Just contact whoever at Thermo gave you access to RawFileReader in the first place. In my case this was jim.shofstahl at thermofisher.com.

closed 2019-04-08 by CarlosHorro

bgruening, 2019-04-09:
Thanks all that have made this possible it is now available at https://usegalaxy.eu/root?tool_id=toolshed.g2.bx.psu.edu/repos/galaxyp/rawtools/rawtools/

kevinkovalchik, 2019-06-06:
Hi @CarlosHorro. I want to let you know we will be releasing 2.0.0 in the near future. It is a major version increment and is going to break the backward compatibility of the CLI. I don't know if you'll want to update the Galaxy Tool wrapper or not. There are a few minor bug fixes in there, and the CLI has changed so we don't have to call "qc" and "parse" separately, but most of the changes are improvements to the GUI, including a tool for visualizing the output data. So if you're using it mostly for creating MGF files it might not be worth updating the galaxy tool wrapper, at least not until v2 has been out there for a while and we iron out issues that pop up.

Anyway, just a heads up!