kevinkovalchik/RawTools #21: Export total ion current TIC from raw file?

soerendip, 2019-01-10:
Hi, 

I wonder whether it is possible to export the total ion current (TIC) from a RAW file with RawTools? Does it happen automatically, or do I have to specify an attribute at the command line? 

I used `RawTools parse -f ../raw/*.raw -p -q -m -x --chro 2T -r TMT10` for a ten plexed TMT labeled MS3 run.

kind regards

kevinkovalchik, 2019-01-10:
Hello,

Yes, that should work fine. I tested the parameters you used on my system and a MS2 TIC was generated.

A couple ideas on why you might be having issues:

* Are you actually using the * wildcard in your command or was it illustrative? RawTools isn't setup to use wildcards, and will crash if they are used. But if you want to process all the files in a directory you can use `-d ../raw` in your example above, and that will find all the raw files in that directory. Wildcards are a nice idea, though. I might try incorporating it into a future release.

* What version of RawTools are you using? The early versions didn't have support for generating chromatograms.

Kevin

soerendip, 2019-01-10:
Hi, 
according to  my bash-history I actually used the wildcard. And, as far as I see, worked fine. I am actually quite new to Proteomics. Do you mind describing where to look? It produced a bunch of files ending with 

 - raw_Matrix.txt
 - raw.mfg
 - raw_Metrics.txt 
 - raw_Ms2_TIC_chromatogram.txt

Is any of the files the file you are talking about? All right, never mind, now I see the TIC in the file. Sometimes, I am blind. Thanks!!!

closed 2019-01-10 by soerendip

kevinkovalchik, 2019-01-10:
Good to know things are working! Also, nice to know wildcards work on your system. Are you using a Mac or Linux? They don't work in my Windows system, but I haven't tried them elsewhere yet. Probably the Linux and Mac consoles are better at translating wildcards into something RawTools likes!

Kevin

kevinkovalchik, 2019-01-10:
> I am using the Windows subsystem for Linux. So, Linux. I am not 100% sure
> that not a command was left out. I might have done it with a for loop.
> Though I check it and tell you. I wonder whether there is a documentation
> of the files that are generated. We wonder whether we have to sum up the
> values in that file to get the TIC. The numbers look quite good by the way.

Oh, I see. Yes, if you sum the values you will have the TIC. That file is a
TIC chromatogram, so TIC for each spectrum. This value might be in the
Metrics file, though I can't remember at the moment.

Don't worry too much about the wild card. I was just curious about it.

The is documentation on the GitHub page. It is in the Wiki section. I think
it is up to date, but let me know if something doesn't line up with the
data output.

Kevin

soerendip, 2019-01-11:
I don't see the Ms2_TIC file listed there. Also not the raw.mfg. The latter is just a conversion from raw to mfg, right? Is it possible to convert it to mzXML with RawTools?
Sören

kevinkovalchik, 2019-01-11:
Oh, sorry. There have been some additions lately that haven't made it into
the Wiki yet.
Yes, the raw.mgf file is a generic format for mass spectra data. It is a
pretty basic format. Each scan within the file has a header, which has some
descriptive information (precursor mass, charge, etc) and a list of peaks
(first column is mass, second column is intensity). Most database search
engines will accept this format as input.
RawTools does not convert to mzXML.