kevinkovalchik/RawTools #24: What is QC standing for?

soerendip, 2019-01-11:
RawTools has the option `qc`. 

> Perform QC operations. Two arguments are required: -d, the directory containing one or more raw
>               files to QC; -q, a directory in which to hold the QC data. -q and -d can be identical if desired.
>               After running for the first time on a given directory, a file called QC.xml will appear in the QC
>               directory. It is an xml file containing the QC data. If new files are copied or acquired into the
>               -d directory, QC may be run again and the new files will be added to the QC data.

Does it stand for quality control?

kevinkovalchik, 2019-01-11:
Yes, it stands for quality control. I see the help is missing some information. An XML file is generated, but that is mostly for storing the data for application use. A CSV file is also created which contains the data in a more accessible format.
The idea is you will use it with a series of files, most likely all the files from a run or a multi-run experiment. You can then visualize the performance metrics generated for all the files to see if and how these metrics are changing with time. We have an online app for viewing this data (https://rawtoolsqcdv.bcgsc.ca/), or it can easily be viewed in R, python, or excel.

kevinkovalchik, 2019-01-11:
You would need to pass the directories one by one. But you can keep referencing the same QC data directory (the -q argument), so the data will get added to the single xml and csv files.

closed 2019-01-24 by soerendip