kevinkovalchik/RawTools #26: Problems generating mgf file with MS2 data.

soerendip, 2019-01-24:
**Describe the bug**
I have MS2 data from a bacterial sample. 

I ran `RawTools parse -f 1_A.raw -m` and get the following output:

> Processing: 1_A.raw
> 
> Indexing linked scan events: 100%                                                                                    
>                                                                                                                      
> 
> Unhandled Exception:
> System.MissingMethodException: Method 'Array.Empty' not found.
>   at RawTools.Program.DoStuff (RawTools.ArgumentParser.ParseOptions opts) <0x41315410 + 0x00f7f> in <filename unknown>:0 
>   at RawTools.Program+<>c.<Main>b__0_0 (RawTools.ArgumentParser.ParseOptions opts) <0x413153e0 + 0x00013> in <filename unknown>:0 
>   at CommandLine.ParserResultExtensions.WithParsed[T] (CommandLine.ParserResult`1 result, System.Action`1 action) <0x413152c0 + 0x000d2> in <filename unknown>:0 
>   at RawTools.Program.Main (System.String[] args) <0x412a1f70 + 0x00337> in <filename unknown>:0 
> [ERROR] FATAL UNHANDLED EXCEPTION: System.MissingMethodException: Method 'Array.Empty' not found.
>   at RawTools.Program.DoStuff (RawTools.ArgumentParser.ParseOptions opts) <0x41315410 + 0x00f7f> in <filename unknown>:0 
>   at RawTools.Program+<>c.<Main>b__0_0 (RawTools.ArgumentParser.ParseOptions opts) <0x413153e0 + 0x00013> in <filename unknown>:0 
>   at CommandLine.ParserResultExtensions.WithParsed[T] (CommandLine.ParserResult`1 result, System.Action`1 action) <0x413152c0 + 0x000d2> in <filename unknown>:0 
>   at RawTools.Program.Main (System.String[] args) <0x412a1f70 + 0x00337> in <filename unknown>:0 
> 

**Description of raw file**
It is MS2 data from an Orbitrap. The file is ~2GB in size.

kevinkovalchik, 2019-01-25:
So this happens when you use `RawTools parse -f 1_A.raw -m` but not when you use `RawTools parse -f 1_A.raw -p`? Which version of RawTools are you using?

kevinkovalchik, 2019-01-25:
Also, does this happen for all your raw files or just this one?

soerendip, 2019-01-29:
Apparently, this was an error related to `mono`. I installed version `5.14.0.177`. Now, the mgf files are generated. With `RawTools parse -f file.raw -m`.

closed 2019-01-29 by soerendip

kevinkovalchik, 2019-01-29:
Okay. Thanks for letting us know!