kevinkovalchik/RawTools #29: Error during quantification processing

chrishuges, 2019-02-04:
**Describe the bug**
Issue is from a user who has contacted me directly. Some files appear to give an error during processing of quantification data. Appears to be using RawTools 1.4.1 on Windows and processing TMT data. 

**Description of raw file**
More details needed here. 

**Command line arguments**
RawTools parse -f C:\myfile.raw -q -r TMT10

**Command line output**
See screenshots.

```
[Paste output here, between the triple quotes]
```

**Screenshots**
[PastedGraphic-1.pdf](https://github.com/kevinkovalchik/RawTools/files/2828617/PastedGraphic-1.pdf)

**Desktop & RawTools information:**
 - OS: Windows
- RawTools version: 1.4.1

**Additional context**
Add any other context about the problem here.

chrishuges, 2019-02-05:
Received a raw file from the user. Processing on our Linux CentOS system using RawTools 1.4.1.

For the metrics output:

```
-bash-4.2$ mono RawTools.exe parse -f /projects/ptx_results/OtherDataSets/RawTools_test-data/170120_JMK_EKS_TMT10_MZM_5w_vs_12w_16frHpH_fr2.raw -x

Processing: /projects/ptx_results/OtherDataSets/RawTools_test-data/170120_JMK_EKS_TMT10_MZM_5w_vs_12w_16frHpH_fr2.raw

Determing MS analysis order... Done!

Extracting scan indices: 100%                                                                                                                                                                             

Checking for orphaned scans: 100%                                                                                                                                                                         

================ Scan indexing report ================

Total scans in file: 87512

Scans linked: 87512

Orphan scans:

None!

All scans accounted for!

======================================================

Extracting reaction events: 100%                                                                                                                                                                          
Extracting scan data: 100%                                                                                                                                                                                
Extracting trailer extras: 100%                                                                                                                                                                           
Extracting precursor masses: 100%                                                                                                                                                                         
Extracting retention times: 100%                                                                                                                                                                          
Analyzing precursor peaks: 100%                                                                                                                                                                           
                                                                                                                                                                                                          
Calculating meta data

  MS1 isolation interference

  MS2 scan cycle density

  Ion injection time

  Duty cycle

  Intensity distribution

  Summed intensities

Calculating metrics

Elapsed time: 80.07 s

Time to process all 1 files: 00:01:20.0738932

-bash-4.2$ 

```
<br>

For the scan matrix output:

```
-bash-4.2$ mono RawTools.exe parse -f /projects/ptx_results/OtherDataSets/RawTools_test-data/170120_JMK_EKS_TMT10_MZM_5w_vs_12w_16frHpH_fr2.raw -p

Processing: /projects/ptx_results/OtherDataSets/RawTools_test-data/170120_JMK_EKS_TMT10_MZM_5w_vs_12w_16frHpH_fr2.raw

Determing MS analysis order... Done!
Extracting scan indices: 100%                                                                                                                                                                             
[... 88 lines cut ...]

  Intensity distribution

  Summed intensities
Quantifying reporter ions: 40%                                                                                                                                                                            
Unhandled Exception:                                                                                                                                                                                      
System.IndexOutOfRangeException: Index was outside the bounds of the array.
  at RawTools.Algorithms.QuantifyReporters.QuantifyOneScan (RawTools.Data.Containers.CentroidStreamData centroidStream, System.String labelingReagent) [0x00102] in <28a57e78d24f4590bf92e34907809bad>:0 
  at RawTools.Algorithms.Quantification.Quantify (RawTools.Data.Collections.CentroidStreamCollection centroidScans, RawTools.Data.Collections.SegmentScanCollection segmentScans, RawTools.WorkFlows.WorkflowParameters parameters, RawTools.Data.Containers.MethodDataContainer methodData, RawTools.Data.Containers.ScanIndex index) [0x00068] in <28a57e78d24f4590bf92e34907809bad>:0 
  at RawTools.WorkFlows.WorkFlowsDDA.ParseDDA (ThermoFisher.CommonCore.Data.Interfaces.IRawFileThreadManager rawFileThreadManager, RawTools.WorkFlows.WorkflowParameters parameters) [0x0012a] in <28a57e78d24f4590bf92e34907809bad>:0 
  at RawTools.Program.DoStuff (RawTools.ArgumentParser.ParseOptions opts) [0x0046b] in <28a57e78d24f4590bf92e34907809bad>:0 
  at RawTools.Program+<>c.<Main>b__0_0 (RawTools.ArgumentParser.ParseOptions opts) [0x00000] in <28a57e78d24f4590bf92e34907809bad>:0 
  at CommandLine.ParserResultExtensions.WithParsed[T] (CommandLine.ParserResult`1[T] result, System.Action`1[T] action) [0x0001e] in <a8f20d44a846452b83e01ced3f113eaa>:0 
  at RawTools.Program.Main (System.String[] args) [0x000d3] in <28a57e78d24f4590bf92e34907809bad>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.IndexOutOfRangeException: Index was outside the bounds of the array.
  at RawTools.Algorithms.QuantifyReporters.QuantifyOneScan (RawTools.Data.Containers.CentroidStreamData centroidStream, System.String labelingReagent) [0x00102] in <28a57e78d24f4590bf92e34907809bad>:0 
[... cut ...]
  at RawTools.Program.DoStuff (RawTools.ArgumentParser.ParseOptions opts) [0x0046b] in <28a57e78d24f4590bf92e34907809bad>:0 
  at RawTools.Program+<>c.<Main>b__0_0 (RawTools.ArgumentParser.ParseOptions opts) [0x00000] in <28a57e78d24f4590bf92e34907809bad>:0 
  at CommandLine.ParserResultExtensions.WithParsed[T] (CommandLine.ParserResult`1[T] result, System.Action`1[T] action) [0x0001e] in <a8f20d44a846452b83e01ced3f113eaa>:0 
  at RawTools.Program.Main (System.String[] args) [0x000d3] in <28a57e78d24f4590bf92e34907809bad>:0 
-bash-4.2$ 

```

chrishuges, 2019-02-05:
Lastly, RawTools 1.3.3 will process the file fine independent of the labeling reagent you specify:

```
-bash-4.2$ mono RawTools.exe parse -f /projects/ptx_results/OtherDataSets/RawTools_test-data/170120_JMK_EKS_TMT10_MZM_5w_vs_12w_16frHpH_fr2.raw -q -r TMT10

Processing: /projects/ptx_results/OtherDataSets/RawTools_test-data/170120_JMK_EKS_TMT10_MZM_5w_vs_12w_16frHpH_fr2.raw

Indexing linked scan events: 100%                                                                                                                                                                         
Extracting raw data: 100%                                                                                                                                                                                 
Analyzing precursor peaks: 100%                                                                                                                                                                           
Quantifying reporter ions: 100%                                                                                                                                                                           
Formatting scan meta data: 100%                                                                                                                                                                           
Writing matrix to disk: 100%                                                                                                                                                                              
                                                                                                                                                                                                          

Elapsed time: 55.66 s

Time to process all 1 files: 00:00:55.6618315

-bash-4.2$ 

```

kevinkovalchik, 2019-02-06:
This is fixed in [v1.4.2](https://github.com/kevinkovalchik/RawTools/releases/tag/1.4.2).

closed 2019-02-06 by kevinkovalchik