kevinkovalchik/RawTools #30: R-Shiny 'to' must be a finite number

soerendip, 2019-02-18:
I am trying to open a  _Matrix file in the Rshiny app. the _Matrix file is fine and was generated with the –p comment from an MS2 / unlabelled dataset using RawTools 1.4.2. 
However, the Rshiny app gives me this error message: 

> “'to' must be a finite number”

Would you know if this is an problem with my file or within Rshiny?
 
thanks
Anne

chrishuges, 2019-02-19:
Hi,

This is a problem with the Shiny app. Nothing on your end. In RawTools 1.4.2 a change was made to a column name that results in the plots from working. We have sent a fix to our IT team, but they haven't pushed it to the web yet. I will post here when they do.

Cheers,
Chris

chrishuges, 2019-02-19:
I put the updated version of the Shiny application [here](https://github.com/kevinkovalchik/RawTools/blob/master/documentation/manuscript/RawTools_RShiny_Application/RawTools_Quality-Control_ShinyApp_v20190218.R), if you want to use it locally in the meantime. 

Cheers,
Chris

soerendip, 2019-02-19:
Thank you Chris!

chrishuges, 2019-02-19:
Hi,

The Shiny application should be updated on the web now. Try it out and let me know if you have any issues.

Cheers,
Chris

soerendip, 2019-02-20:
It seems to be working.

closed 2019-02-20 by soerendip