kevinkovalchik/RawTools #31: Input string was not in a correct format

armadillocommander, 2019-02-19:
**Describe the bug**
Tried to run QC on a bunch of shotgun proteomics raw data (Top N DDA acquisition).

**Description of raw file**
Top N DDA acquisition for 60 min, ~600 MB for each.

**Command line arguments**
RawTools.exe qc -d g:\path_to_raw_data\ -q g:\path_to_raw_data\QC

**Command line output**

```
Processing g:\path_to_raw_data\A1_rep3.raw

Unhandled Exception: System.FormatException: Input string was not in a correct format.
   at System.Text.StringBuilder.FormatError()
   at System.Text.StringBuilder.AppendFormatHelper(IFormatProvider provider, String format, ParamsArray args)
   at System.String.FormatHelper(IFormatProvider provider, String format, ParamsArray args)
   at System.IO.TextWriter.WriteLine(String format, Object arg0)
   at System.IO.TextWriter.SyncTextWriter.WriteLine(String format, Object arg0)
   at System.Console.WriteLine(String format, Object arg0)
   at RawTools.QC.QcWorkflow.DoQc(WorkflowParameters parameters) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\QC\QcWorkflow.cs:line 96
   at RawTools.Program.DoStuff(QcOptions opts) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\Program.cs:line 357
   at CommandLine.ParserResultExtensions.WithParsed[T](ParserResult`1 result, Action`1 action)
   at RawTools.Program.Main(String[] args) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\Program.cs:line 69
```

**Desktop & RawTools information:**
 - OS: Windows 10 Version 1803 (Build 17134.590)
- RawTools version: 1.4.2

kevinkovalchik, 2019-02-19:
Hello,

Is it possible that one or more of that there are renamed duplicate raw files? RawTools determines the ordering of raw files for QC based on creation date and time, and when it goes through the list of files if it comes across one that has already been entered it is supposed to print out a message to the console and skip that file. There was an error in that print statement, which is what you saw.

However, if your files are all unique then it probably means that copying or moving the files is messing up the date-time information and I need to come up with a better way to do it.

Kevin

armadillocommander, 2019-02-19:
@kevinkovalchik You are right! There was a duplicate file was directly copied from another file. Thanks!

closed 2019-02-19 by armadillocommander

kevinkovalchik, 2019-02-19:
Great, thanks for letting me know! I will fix the bug so this doesn't throw an error in the future.