kevinkovalchik/RawTools #37: Extracting & plotting peptide/protein results from X!tandem search

roberthardt, 2019-05-07:
**Is your feature request related to a problem? Please describe.**
Currently just a few metrics are taken from the X!tandem search and plottet by the Shiny app. 

**Describe the solution you'd like**
It would be fantastic to plot also things like the peptide scores, average peptide scores, proteins cores, protein sequence coverage. Since we always run the same sample (BSA digest) for our QC, we are interested how good the identification was in each run.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kevinkovalchik, 2019-05-07:
Thanks for the suggestion. I don't think the "preview" search we do would
be adequate for things like protein coverage, so those metrics would
probably need to trigger a full search, which is not a problem. It'll just
take longer. I'll talk with Chris about it and get back to you to
discuss further.

Kevin

kevinkovalchik, 2019-05-09:
@roberthardt We don't want to get too specific with the QC metrics, as we are more interested in observing consistency rather than monitoring a specific QC cutoff. However, you are right that there is some room for additional features here. I am going to look into adding these metrics to the QC data:

- Median peptide score
- Decoy score cutoff value
- Number of PSMs
- Number of unique peptides

roberthardt, 2019-05-09:
Hi Kevin,

Thanks for your quick reply and the future implementation of the additional database search derived metrics. Having those additional informations in the QC table would allow us assessing our LC-MS systems in a fast and convenient way. Currently we do those things manually, which is very time consuming and tedious. 

Best regards

Robert 

--------------------------
Dr. Robert Hardt
Emser Str. 32
65195 Wiesbaden 
Tel.: +49-176-61573795

kevinkovalchik, 2019-07-03:
Hi @roberthardt,

v2.0.0 was just released today and includes the metrics mentioned above.

Let us know if there are other improvements you'd like to see!

Kevin

closed 2019-07-03 by kevinkovalchik