kevinkovalchik/RawTools #38: Additional useful metric/setting suggestions

roberthardt, 2019-05-28:
**Is your feature request related to a problem? Please describe.**
no

**Describe the solution you'd like**
1. Would it possible to also state the average number of datapoints over the MS1-peaks? This is usually quite helpful to assess how good your quantification could be.
2. I browsed through some of my data using the Shiny-app. Would it be possible to add a file/lc-ms run filter function in this, which directly affects the plots? With this one could then simply get plots for a subgroup of raw files.

**Describe alternatives you've considered**
---

**Additional context**
---

Thanks again for the great tool you provided for the community.

Best

Robert

roberthardt, 2019-05-28:
> Hi Robert,
>
> Thanks again for these suggestions. I've got questions on each point:
>
> Would the data points represent MS1 scans, or MS2/MS3 scans? i.e. is the metric getting at the number of DDA cycles happening across the peak profile or the number of times a peak triggers an MS2 event?
>
> This could be a little tricky. Would you want to simply select a range or list based on run number? e.g. "show me runs 20-30". Or would you like to be able to do some more advanced filtering based on filenames? e.g. using regular expressions
>
> In regards to number 2, I've been working on a standalone visualization app to release alongside RawTools. Hopefully I'll be able to release this in some form later this week. As of now, it doesn't have anything like you requested, but I'll work on adding it in the future.
>
> Take care,
> Kevin

Hi Kevin,

Thanks again for the immediate reply. 

Regarding point 1:
I’m interested in first, the number of dda cycles. I know there is the cycle time and I could easily calculate that by taking the peak width and dividing it by the cycle time, but I rather have a separate metric. 

2. Here I’m interested in being able to do the later i.e. more advanced filtering. I for example usually run low complex (BSA) as well as high complex (HeLa) samples on our mass specs. Filtering the results for either one of those would be great. Here I could work with different QC csv files for each sample type, but a filtering option would be the more elegant solution. 

I am also excited about the future new visualization tool. Trying to finish it until ASMS?😉

Keep up the good work

Best Robert 

--------------------------
Dr. Robert Hardt
Emser Str. 32
65195 Wiesbaden 
Tel.: +49-176-61573795

kevinkovalchik, 2019-05-28:
Haha, no unfortunately and oddly the CSC conference and ASMS are concurrent this year and I'm presenting at CSC.

Okay, those should be easy to incorporate. For 2, a simple text search will definitely work (e.g. filter for "bsa" in the filename). I can enable regular expressions if something fancier is desired, but I imagine that would be overkill.

kevinkovalchik, 2019-07-03:
Hi @roberthardt,

These changes have been made in v2.0.0. For the filtering, we didn't go as far as allowing for regex expressions, but there is a simple text filter for raw file names. For example, if you filter for "HeLa", it will display data only for raw files with "HeLa" as part of the filename. Let me know more flexible filtering would be helpful.

Kevin

closed 2019-07-03 by kevinkovalchik

roberthardt, 2019-07-04:
Hi Kevin,

thanks for the new version of RawTools and implementing my suggestions. I
will check it out right now. :-)

Best

Robert

roberthardt, 2019-07-04:
Hi Kevin,

The visualization tool always crashes when I'm trying to load in any result
created by RawTools. Did I necessarily have to install Thermo's
new RawFileReader first?

best

Robert

kevinkovalchik, 2019-07-04:
Hi Robert,

Sorry about that. Can you tell me what operating system you are using? And
is an error message popping up or is the tool just closing when you try to
open something?

Kevin

kevinkovalchik, 2019-07-04:
Oh, I see. I can't even run it myself... well that's embarrassing. I must
have forgotten to copy some settings in the release build. I tested it, but
it must have still been running the debug version. I'll see if I can get
this fixed ASAP!

kevinkovalchik, 2019-07-04:
Okay, I fixed it! Please try v2.0.1.

Sorry again, I should have caught that before the release.

Kevin