kevinkovalchik/RawTools #40: X!tandem did not work

jefftong2019, 2019-06-26:
**Describe the bug**
I select X!tandem option for search and select right fold with X!tandem.exe

**Description of raw file**
Used RawToolGUI.exe to run TMT labeled samples with MS2 quantification or MS3 quantification (different files).  Program runs very fast, but the identificationRate for MS2 is 0.025 and it seems not right.  

**Command line arguments**
I did not run command line, but the GUI interface.

**Command line output**
No output line and I do get QC.csv file

**Screenshots**
[image]

**Desktop & RawTools information:**
 - OS: [Window10]
- RawTools version: new version from Dr.Chris Hughes

**Additional context**
Add any other context about the problem here.

kevinkovalchik, 2019-06-26:
Thanks for reporting the problem. I see you have a version that Chris gave you. When did you get it from him? There was an issue with the peptide modifications in a recent test version of the GUI which might explain the problem. It is fixed in the latest test version I gave him (June 24, 2019).

jefftong2019, 2019-06-26:
Hi Kevin,

Thanks for your quick reply.  We got May 29, 2019 version.  
Can I down load new version online? or could you send me a linker in dropbox?
Thanks a lot.
-Jiefei

kevinkovalchik, 2019-06-26:
Hi Jiefie,

Sure here's the latest version: [RawTools_20190626.zip](https://github.com/kevinkovalchik/RawTools/files/3331572/RawTools_20190626.zip)

Let me know how it goes!

Kevin

jefftong2019, 2019-06-26:
Hi Kevin,

Thanks.  got it.  Will let you how it goes. 
-Jiefei

chrishuges, 2019-07-03:
Have communicated with this user and all is working well now.

closed 2019-07-03 by chrishuges

jefftong2019, 2019-07-03:
Hi Kevin,

I am sorry forgot to let you know.  The new version works for me now. 
Thanks a lot.

-Jiefei