kevinkovalchik/RawTools #42: RawTools v2.0.1 error on Linux

chrishuges, 2019-07-08:
**Describe the bug**
Running RawTools on Linux (CentOS 7) I get an error in the command line. 

**Description of raw file**
None.

**Command line arguments**

```
mono /projects/ptx_analysis/chughes/software/RawTools/rawtools201/RawTools.exe
```

**Command line output**

```
Unhandled Exception:
System.ArgumentOutOfRangeException: Value must be positive and below the buffer height.
Parameter name: top
  at System.TermInfoDriver.SetCursorPosition (System.Int32 left, System.Int32 top) [0x0003e] in <135e7c260ae04ab1a5538976c2d8cf31>:0 
  at System.TermInfoDriver.set_CursorTop (System.Int32 value) [0x00015] in <135e7c260ae04ab1a5538976c2d8cf31>:0 
  at System.ConsoleDriver.set_CursorTop (System.Int32 value) [0x00000] in <135e7c260ae04ab1a5538976c2d8cf31>:0 
  at System.Console.set_CursorTop (System.Int32 value) [0x00000] in <135e7c260ae04ab1a5538976c2d8cf31>:0 
  at CLParser.HelpText.WriteLine (System.Char line, System.String text) [0x00005] in <2e4e39d756884c88af8586caf674e0cb>:0 
  at CLParser.HelpText.Help (CLParser.ClParser parser) [0x00018] in <2e4e39d756884c88af8586caf674e0cb>:0 
  at CLParser.ClParser.Parse (System.String[] args) [0x00040] in <2e4e39d756884c88af8586caf674e0cb>:0 
  at RawTools.Program.Main (System.String[] args) [0x000ac] in <e304032c5acb4ff1ad322c870466723d>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.ArgumentOutOfRangeException: Value must be positive and below the buffer height.
Parameter name: top
  at System.TermInfoDriver.SetCursorPosition (System.Int32 left, System.Int32 top) [0x0003e] in <135e7c260ae04ab1a5538976c2d8cf31>:0 
  at System.TermInfoDriver.set_CursorTop (System.Int32 value) [0x00015] in <135e7c260ae04ab1a5538976c2d8cf31>:0 
  at System.ConsoleDriver.set_CursorTop (System.Int32 value) [0x00000] in <135e7c260ae04ab1a5538976c2d8cf31>:0 
  at System.Console.set_CursorTop (System.Int32 value) [0x00000] in <135e7c260ae04ab1a5538976c2d8cf31>:0 
  at CLParser.HelpText.WriteLine (System.Char line, System.String text) [0x00005] in <2e4e39d756884c88af8586caf674e0cb>:0 
  at CLParser.HelpText.Help (CLParser.ClParser parser) [0x00018] in <2e4e39d756884c88af8586caf674e0cb>:0 
  at CLParser.ClParser.Parse (System.String[] args) [0x00040] in <2e4e39d756884c88af8586caf674e0cb>:0 
  at RawTools.Program.Main (System.String[] args) [0x000ac] in <e304032c5acb4ff1ad322c870466723d>:0 

```

**Screenshots**
None.

**Desktop & RawTools information:**
 - OS: CentOS 7
- RawTools version: 2.0.1

**Additional context**
Add any other context about the problem here.

chrishuges, 2019-07-08:
On MacOS, it makes it a bit farther depending on the command:

Command

```
mono RawTools.exe
```

Output

```
========== ABOUT ==============================================================================================================================================================================================================

  Welcome to the RawTools page! RawTools is an open-source and freely available package designed to perform scan data parsing and quantification, and quality control analysis of Thermo Orbitrap raw mass spectrometer files. 
  RawTools uses the Thermo RawFileReader library (Copyright © 2016 by Thermo Fisher Scientific, Inc. All rights reserved). RawTools is fully compatible with Windows, Linux, and MacOS operating systems. 

========== PARAMETERS HELP ====================================================================================================================================================================================================

  Below you will find information on all the parameters available to this application. The Parameter field is a (hopefully) descriptive name of the parameter which might be referenced when encountering an error. Short form 
  and Long form are how you invoke the parameters on the command line. You can use either form. Required indicates whether the application requires the parameter have a value in order to run. 

  INFO                                DESCRIPTION
  ----                                -----------

  Parameter: RawFiles                 Indicates input file(s) to be processed, separated by a space if there are multiple files. Must be Thermo .raw files. You must use either -f or -d to indicate the file(s) to process. 
  Short form: -f 
  Long form: -files 
  Required: False 

  Parameter: RawFileDirectory         Indicates directory to be processed. Files other than .raw files will be ignored. You must use either -d or -f to indicate the file(s) to process. 
  Short form: -d 
  Long form: -directory 
  Required: False 

  Parameter: SearchSubdirectories     Indicate that, in addition to the directory specified by -d, search all subdirectories for .raw files. 
  Short form: -s 
  Long form: -searchsubdirectories 
  Required: False 

  Parameter: QcDirectory              Indicates you want to perform QC. Must be followed by the tath to the directory containing (or to contain) the QC data file (called QC.xml). 
  Short form: -qc 
  Long form: -qcdirectory 
  Required: False 

  Parameter: Parse                    Parses raw file meta and scan data and writes the output to a tab-delimited text file. Typically either this output or the quant output (-q) is used unless your aim is to simply create 
  Short form: -p                      an MGF or to observe broad metrics using -x. 
  Long form: -parse 
  Required: False 

  Parameter: Quant                    Similar to parse (-p), but also quantifies reporter ions and write results to output matrix. Use of this flag requires you also specify the reagents used for isobaric labeling with the 
  Short form: -q                      -r argument (e.g. -r TMT10) 
  Long form: -quant 
  Required: False 

  Parameter: LabelingReagent          Required for reporter ion quantification. Reagents used to label peptides, required if using quant option. Available options are: {TMT0, TMT2, TMT6, TMT10, TMT11, iTRAQ4, iTRAQ8}. 
  Short form: -r 
  Long form: -labellingreagent 
  Required: False 

  Parameter: UnlabeledQuant           Calculate areas of precursor peaks and writes them to the parse or quant file (ParentPeakArea column). This option is to be used in combination with -p or -q. 
  Short form: -u 
  Long form: -unlabeledquant 
  Required: False 

  Parameter: WriteMGF                 Writes a standard MGF file. To specify a mass cutoff use the -c argument. 
  Short form: -m 
  Long form: -mgf 
  Required: False 

  Parameter: MgfMassCutoff 
[... 32 lines cut ...]
Output

```
Unhandled Exception:
System.ArgumentOutOfRangeException: Value must be positive and below the buffer height.
Parameter name: top
  at System.TermInfoDriver.SetCursorPosition (System.Int32 left, System.Int32 top) [0x0003e] in <4ce69280878a4ebdbc54175071d7ecdf>:0 
  at System.TermInfoDriver.set_CursorTop (System.Int32 value) [0x00015] in <4ce69280878a4ebdbc54175071d7ecdf>:0 
  at System.ConsoleDriver.set_CursorTop (System.Int32 value) [0x00000] in <4ce69280878a4ebdbc54175071d7ecdf>:0 
  at System.Console.set_CursorTop (System.Int32 value) [0x00000] in <4ce69280878a4ebdbc54175071d7ecdf>:0 
  at CLParser.HelpText.WriteLine (System.Char line, System.String text) [0x00005] in <2e4e39d756884c88af8586caf674e0cb>:0 
  at CLParser.HelpText.Help (CLParser.ClParser parser) [0x00018] in <2e4e39d756884c88af8586caf674e0cb>:0 
  at CLParser.ClParser.Parse (System.String[] args) [0x00040] in <2e4e39d756884c88af8586caf674e0cb>:0 
  at RawTools.Program.Main (System.String[] args) [0x000ac] in <e304032c5acb4ff1ad322c870466723d>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.ArgumentOutOfRangeException: Value must be positive and below the buffer height.
Parameter name: top
  at System.TermInfoDriver.SetCursorPosition (System.Int32 left, System.Int32 top) [0x0003e] in <4ce69280878a4ebdbc54175071d7ecdf>:0 
  at System.TermInfoDriver.set_CursorTop (System.Int32 value) [0x00015] in <4ce69280878a4ebdbc54175071d7ecdf>:0 
  at System.ConsoleDriver.set_CursorTop (System.Int32 value) [0x00000] in <4ce69280878a4ebdbc54175071d7ecdf>:0 
[... cut ...] Int32 value) [0x00000] in <4ce69280878a4ebdbc54175071d7ecdf>:0 
  at CLParser.HelpText.WriteLine (System.Char line, System.String text) [0x00005] in <2e4e39d756884c88af8586caf674e0cb>:0 
  at CLParser.HelpText.Help (CLParser.ClParser parser) [0x00018] in <2e4e39d756884c88af8586caf674e0cb>:0 
  at CLParser.ClParser.Parse (System.String[] args) [0x00040] in <2e4e39d756884c88af8586caf674e0cb>:0 
  at RawTools.Program.Main (System.String[] args) [0x000ac] in <e304032c5acb4ff1ad322c870466723d>:0 
```

kevinkovalchik, 2019-07-08:
Ah. The windows console seems to handle cursor position differently from Linux and OSX. There doesn't seem to be an easy way around this that I can see, so I'll change how I'm displaying the help text to a way that doesn't involve manipulating the cursor.

kevinkovalchik, 2019-07-08:
This has been fixed and will be released in v2.0.2.

closed 2019-07-08 by kevinkovalchik