kevinkovalchik/RawTools #46: Can't parse my Raw File

ncarrut, 2019-08-21:
**Describe the bug**
rawTools failse to parse my QC run. May be the same issue as:  https://github.com/kevinkovalchik/RawTools/issues/2

**Description of raw file**
100 MB raw file from a 25 minute run.  It's a yeast digest with Pierce RT peptides spiked in.  Analysis is mixed DDA and targeted for the RT peptides.  

**Command line arguments**
RawTools.exe -f Yeast_20181109170737.raw -x

**Command line output**
Please paste the entire output from the command line, if applicable (i.e. everything that appeared after you hit enter until things stopped working).
```
        Scan: 20738, MSOrder: Ms2
        Scan: 20739, MSOrder: Ms2
        Scan: 20740, MSOrder: Ms2
        Scan: 20741, MSOrder: Ms2
        Scan: 20742, MSOrder: Ms2
        Scan: 20743, MSOrder: Ms2
        Scan: 20744, MSOrder: Ms2

The above scans will not be present in the output data. You should
manually check them to ensure they are not critical to you analysis.

All scans accounted for!
======================================================

Extracting reaction events: 100%

Unhandled Exception: System.IndexOutOfRangeException: Index was outside the bounds of the array.
   at RawTools.Algorithms.ExtractData.Extract.MethodData(IRawDataPlus rawFile, ScanIndex index) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\Algorithms\Extraction\ExtractData.cs:line 763
   at RawTools.WorkFlows.WorkFlowsDDA.UniversalDDA(IRawFileThreadManager rawFileThreadManager, WorkflowParameters parameters, QcDataCollection qcDataCollection) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\WorkFlows\DDA.cs:line 80
   at RawTools.Program.Run(Dictionary`2 opts) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\Program.cs:line 221
   at RawTools.Program.Main(String[] args) in C:\Users\Kevin\Documents\GSC\Projects\RawTools\RawTools\Program.cs:line 73
```

**Desktop & RawTools information:**
 - OS: windows 10
- RawTools version: 2.0.2

**Additional context**
Add any other context about the problem here.

armadillocommander, 2019-10-10:
I have the same issue on Windows 10.

chrishuges, 2019-10-10:
I assume this is because the data is a mix of targeted and dda. I imagine the targeted data is causing the issue. Is it possible to share the raw file (or any raw file with the same scan type) to help solve the issue?