kevinkovalchik/RawTools #50: System.MissingMethodException: Method 'Array.Empty' not found

eltonjrv, 2020-04-06:
**Describe the bug**
Trying to run Rawtools for quality control analysis along with a database search on TMT10 raw data which has an empty channel.

**Description of raw file**
MS2 - TMT10.

**Command line arguments**
$ mono /usr/local/RawTools/RawTools.exe -d . -qc gluc-qc_Btau_trembl -db ../Btaurus-uniprot-trembl.fasta -fmods 57.0214@C -vmods 15.9949@M -N 1000 -X /usr/local/RawTools/tandem-linux-15-12-15-2/bin/

**Command line output**
Processing: /run/media/elton/TOSHIBA EXT/TMTrawData/glucose/./130619_Fraction_1.raw

Determing MS analysis order... Done!
Extracting scan indices: 100%                                                                                                                         
Checking for orphaned scans: 100%                                                                                                                     
                                                                                                                                                      

================ Scan indexing report ================

Total scans in file: 62959

Scans linked: 62959

Orphan scans:

None!

All scans accounted for!

======================================================

Extracting reaction events: 100%                                                                                                                      
Extracting scan data: 100%                                                                                                                            
Extracting trailer extras: 100%                                                                                                                       
Extracting precursor masses: 100%                                                                                                                     
Extracting retention times: 100%                                                                                                                      
Analyzing precursor peaks: 100%                                                                                                                       
                                                                                                                                                      
Calculating meta data

  MS1 isolation interference

  MS2 scan cycle density

  Ion injection time

  Duty cycle

  Intensity distribution

  Summed intensities

Calculating metrics

Unhandled Exception:
System.MissingMethodException: Method 'Array.Empty' not found.
  at RawTools.QC.Search.WriteSearchMGF (RawTools.WorkFlows.WorkflowParameters parameters, RawTools.Data.Collections.CentroidStreamCollection centroids, RawTools.Data.Collections.SegmentScanCollection segments, RawTools.Data.Collections.RetentionTimeCollection retentionTimes, RawTools.Data.Collections.PrecursorMassCollection precursorMasses, RawTools.Data.Collections.PrecursorScanCollection precursorScans, RawTools.Data.Collections.TrailerExtraCollection trailerExtras, RawTools.Data.Containers.MethodDataContainer methodData, RawTools.Data.Containers.ScanIndex index, System.String rawFileName, Boolean fixedScans) <0x4062dcb0 + 0x000d7> in <filename unknown>:0 
  at RawTools.WorkFlows.WorkFlowsDDA.UniversalDDA (IRawFileThreadManager rawFileThreadManager, RawTools.WorkFlows.WorkflowParameters parameters, RawTools.QC.QcDataCollection qcDataCollection) <0x405a7000 + 0x0099b> in <filename unknown>:0 
  at RawTools.Program.Run (System.Collections.Generic.Dictionary`2 opts) <0x40569650 + 0x00cb7> in <filename unknown>:0 
  at RawTools.Program.Main (System.String[] args) <0x404e3e30 + 0x0021f> in <filename unknown>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.MissingMethodException: Method 'Array.Empty' not found.
  at RawTools.QC.Search.WriteSearchMGF (RawTools.WorkFlows.WorkflowParameters parameters, RawTools.Data.Collections.CentroidStreamCollection centroids, RawTools.Data.Collections.SegmentScanCollection segments, RawTools.Data.Collections.RetentionTimeCollection retentionTimes, RawTools.Data.Collections.PrecursorMassCollection precursorMasses, RawTools.Data.Collections.PrecursorScanCollection precursorScans, RawTools.Data.Collections.TrailerExtraCollection trailerExtras, RawTools.Data.Containers.MethodDataContainer methodData, RawTools.Data.Containers.ScanIndex index, System.String rawFileName, Boolean fixedScans) <0x4062dcb0 + 0x000d7> in <filename unknown>:0 
  at RawTools.WorkFlows.WorkFlowsDDA.UniversalDDA (IRawFileThreadManager rawFileThreadManager, RawTools.WorkFlows.WorkflowParameters parameters, RawTools.QC.QcDataCollection qcDataCollection) <0x405a7000 + 0x0099b> in <filename unknown>:0 
  at RawTools.Program.Run (System.Collections.Generic.Dictionary`2 opts) <0x40569650 + 0x00cb7> in <filename unknown>:0 
  at RawTools.Program.Main (System.String[] args) <0x404e3e30 + 0x0021f> in <filename unknown>:0 

```
[Paste output here, between the triple quotes]
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop & RawTools information:**
 - OS: [e.g. iOS]
- RawTools version: [e.g. 1.3.3]

**Additional context**
Add any other context about the problem here.

kevinkovalchik, 2020-04-06:
> Boolean fixedScans) <0x4062dcb0 + 0x000d7> in :0
> RawTools.QC.QcDataCollection qcDataCollection) <0x405a7000 + 0x0099b> in :0
> (System.Collections.Generic.Dictionary2 opts) <0x40569650 + 0x00cb7> in :0

Thanks for reporting this. RawTools has been on the back burner for a
while, but with all that is happening now work has slowed down and I might
have some time to get to some of the open issues.

Best,
Kevin

chrishuges, 2020-04-16:
I can take a look at this as well. Can you try updating to the [latest version](https://github.com/kevinkovalchik/RawTools/releases/tag/2.0.2) of the software and trying this command again?

Thanks,
Chris

eltonjrv, 2020-04-18:
I'm already using the latest 2.0.2 one.
Is there any other suggestion?
Thank, though.

chrishuges, 2020-04-18:
Yes, my mistake. I misread the version. 

If you process this raw file without a search, just in the regular raw tools pipeline, including mgf creation and quant, does it work without error?

eltonjrv, 2020-04-18:
Yes, the regular "RawTools for quantification" command work well on the same data.

chrishuges, 2020-04-18:
I can't seem to reproduce this error on my own system (CentOS7, RawTools 2.0.2) using a command essentially the same as yours (relative paths, same input flags) and a DDA TMT file (MS2 TMT, or MS3 TMT). 

Is it at all possible to share the raw file giving you an issue so I can test it on my system to help track down where the issue might be?

eltonjrv, 2020-04-18:
Thanks for your prompt attention on that, Chirs!
I don't think the PI owner of that data would allow such thing. Sorry!
Did you get on my initial message that it is a TMT 10-plex lacking one channel (we only have 9 samples). Could that be the reason of this issue?

chrishuges, 2020-04-18:
Yes, the TMT empty channel should make no difference. The test runs I did on my system also had empty channels, just to make sure.

Is there anything else you can tell me about the method used to acquire the data? Instrument, acquisition method, sample complexity? Any details you can give would be helpful. 

Having the raw data would help a lot. Just to say, in case this helps, but I have no way of telling anything about the data from just the raw file. Without knowing the TMT sample assignments, what the sample is, or where it is from, it is impossible for me to do anything of use with these data beyond testing it with our tool.

kevinkovalchik, 2020-04-18:
Do you know what version of dotnet or mono you are using? The missing
array.empty method sounds familiar to me, but I don't recall what the
context was...
Anyway, I'll have a look at the code on Monday.

kevinkovalchik, 2020-04-20:
This could possibly be an issue with an old version of Mono. Is it possible to update and try with the new version?

eltonjrv, 2020-04-21:
Hey guys,
Thanks again for your attention and sorry for being out for a while!
I've been working on my Windows partition in the past two days due to some particular collaborations' requirements. Once I reboot my laptop on my Fedora partition, I'll tell you which mono version I'm using.
Kevin might be right regarding an outdated mono version, since I also have an outdated Fedora one.
JFYI: I already circumvented this situation two weeks ago by running Comet on all mgf files, and then writing a PERL script to match scans from Comet results to the ones from your RawTools -q output.
But I'm sure it will definitely be cool to compare those results with your "QC with database search" tool, which I'm aware uses X!Tandem as the search engine.