kevinkovalchik/RawTools #54: Impurity correction

pfangpGlyco, 2020-07-31:
Hello Kevin,
Thanks for developing such a great tool. Its very helpful for my glycopeptide quantification. We have developed a automatic data processing tool for quantitative glycopeptides by integrating RawTools into the tool. Please see the BioRxiv link: https://www.biorxiv.org/content/10.1101/2020.05.20.105726v1
Now we have met a problem with impurity correction. Did you already put this into the current release?

Best regards,
Pan

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

kevinkovalchik, 2020-08-03:
Hi Pan,
Sorry for not getting around to this after your earlier request. RawTools
is unfortunately not part of my paid work at the moment, so it is more of a
hobby right now. Having said that, I like to see it being used in new tools
and I would like to help you out with the quantification issue. Could you
please upload an example of the input format you would like? Ideally the
format will be .csv or .tsv. I will see what I can do in the next couple
weeks.
Related to this, we've had some other interest in having RawTools support
TMTpro. If I get this done for the same release, will the TMTpro support be
of interest for your tool?

Best wishes,
Kevin

pfangpGlyco, 2020-08-04:
Hi Kevin, 
I'm appreciative with your help. Our tool, GlycoBinder is the first and very important tool for quantitative glycoproteomics. If the impurity correction could be added, it would be perfect. Here is the github of GlycoBinder. https://github.com/IvanSilbern/GlycoBinder
[Impurity Correction Facors.xlsx](https://github.com/kevinkovalchik/RawTools/files/5022719/Impurity.Correction.Facors.xlsx)
I have uploaded a impurity correction factor file with .xlsx format as the github dose not support the csv file. Please let me know if you have any questions. 
I'm very happy to know RawTools will support TMTpro, Thus. our GlycoBinder can be furthur expanded to TMTpro. 
Thanks again for your efforts on RawTools.
Best regards,
Pan

kevinkovalchik, 2020-10-06:
Hi Pan,

I wanted to let you know I started working on this today (in the "impurity_correction" branch). Sorry for the delay, it is difficult to find time around my work schedule. It will probably still be a couple weeks before anything is ready.

Best wishes,
Kevin

pfangpGlyco, 2020-10-07:
Dear Kevin,

I'm appreciated with your help and I'm sorry to occupy your own time to work on this. Please take your time. I'm happy to tell you that our manuscript "SugarQuant: a streamlined pipeline for multiplexed quantitative site-specific N-glycoproteomics" has been accepted by nature communiation. If you finish the impurity correction, we will update our Glycobinder, too.  I believe more scientists will use your excellent RawTools. In addition, you asked me whether we are intrested in TMT 16 last time. If you could put this part in the updated RawTools, that would be great.

Thanks very much for your help.

Best regards,

Pan Faang

--
Pan Fang, PhD
Bioanalytical Mass Spectrometry Group,
Max Planck Institute for Biophysical Chemistry
Am Fassberg 11,
D-37077 Goettingen, Germany

Tel: +49-551-201-1477
Fax: +49-551-201-1197
Email: pfang@mpibpc.mpg.de
http://www.mpibpc.mpg.de/urlaub<https://email.gwdg.de/owa/redir.aspx?C=r-C7RUNxboDxgp5lBMnC1_vJ-tnL8HrpTlIN2I7IpC9EE-rkHxbVCA..&URL=http%3a%2f%2fwww.mpibpc.mpg.de%2furlaub>

kevinkovalchik, 2020-10-07:
Hi Pan,

No problem. I enjoy working on it, there just isn't much time to do so! I
should be able to add TMTpro without too much difficulty. The
quantification code is already there, obviously, so I will just need to add
another set of reporter ion possibilities to it.

Kevin

chrishuges, 2020-10-23:
Following up on this to highlight that TMT16 would be a great thing to add right now. Just from looking around a bit, there are surprisingly few tools that support this at the moment.

kevinkovalchik, 2021-02-03:
Dear Pan,

My apologies for taking so long on this. I wanted to let you know I haven't forgotten about it. I'll be testing some functions this week, so if we're lucky I might have something for you later this month.

Kevin