kevinkovalchik/RawTools #56: Running QC on single raw files

5utr, 2020-10-20:
Is there a way to run the QC mode on single files and not on an entire directory? Sometimes one of the raw files will throw an error and halt the execution on the rest of the directory, having the option to run qc with the -f mode would be helpful.
Thank you

kevinkovalchik, 2020-10-22:
This isn't currently possible, but I might be able to implement something. Do you just want a one-time QC run on a single file or list of files? Or do you want to have a running QC database over time, but you want to be able to specify which files to add?

Thanks,
Kevin

5utr, 2020-10-22:
Ideally a list of files, running on the entire directory can take a lot of time and creating ad hoc directory might require moving several GB of raw data.
Thanks!

soerendip, 2021-04-25:
I had the same isse. I specifically had to design my pipeline so that each .RAW file is stored in a separate directory. A workaround could be to store a symbolic link in a separate directory and then run RawToolsQC.

I also build my own parser to create csv output rather than XML.