kevinkovalchik/RawTools #58: System.AggregateException

mtinti, 2021-04-01:
**Describe the bug**
Rawtools -p rise error on raw files while -qc works fine

**Description of raw file**
TMT-10 plex analysed on lumus MS3

**Command line arguments**
mono RawTools.exe -d ~/host/calvin/new_data/036_2021_Platinum_DN -p -o ~/host/calvin/new_data/036_2021_Platinum_DN/qc_output/

**Command line output**
Please paste the entire output from the command line, if applicable (i.e. everything that appeared after you hit enter until things stopped working).
```
Processing: /home/mik/host/calvin/new_data/036_2021_Platinum_DN/DN-Phos-frac-1.raw

Determing MS analysis order... Done!
                                                                                                                                                                                      Extracting scan indices: 23%
Unhandled Exception:
System.AggregateException: One or more errors occurred. (Value must be positive and below the buffer heigh                                                                            t.
Parameter name: top) ---> System.ArgumentOutOfRangeException: Value must be positive and below the buffer                                                                             height.
Parameter name: top
  at System.TermInfoDriver.SetCursorPosition (System.Int32 left, System.Int32 top) [0x0004d] in <12b418a78                                                                            18c4ca0893feeaaf67f1e7f>:0
  at System.ConsoleDriver.SetCursorPosition (System.Int32 left, System.Int32 top) [0x00000] in <12b418a781                                                                            8c4ca0893feeaaf67f1e7f>:0
  at System.Console.SetCursorPosition (System.Int32 left, System.Int32 top) [0x00000] in <12b418a7818c4ca0                                                                            893feeaaf67f1e7f>:0
  at RawTools.Utilities.ConsoleUtils.ClearLastLine () [0x00006] in <651c04d8311845c091809ac79a3ed047>:0
  at RawTools.Utilities.ProgressIndicator.Update () [0x0000f] in <651c04d8311845c091809ac79a3ed047>:0
  at RawTools.Algorithms.ExtractData.Extract+<>c__DisplayClass0_0.<ScanIndicesPrecursorsDependents>b__1 (S                                                                            ystem.Collections.Generic.IEnumerable`1[T] batch) [0x002b8] in <651c04d8311845c091809ac79a3ed047>:0
  at System.Threading.Tasks.Parallel+<PartitionerForEachWorker>c__AnonStorey6`2[TSource,TLocal].<>m__1 (Sy                                                                            stem.Collections.IEnumerator& partitionState, System.Int32 timeout, System.Boolean& replicationDelegateYie                                                                            ldedBeforeCompletion) [0x0013f] in <12b418a7818c4ca0893feeaaf67f1e7f>:0
--- End of stack trace from previous location where exception was thrown ---

  at System.Threading.Tasks.Parallel+<PartitionerForEachWorker>c__AnonStorey6`2[TSource,TLocal].<>m__1 (Sy                                                                            stem.Collections.IEnumerator& partitionState, System.Int32 timeout, System.Boolean& replicationDelegateYie                                                                            ldedBeforeCompletion) [0x002f6] in <12b418a7818c4ca0893feeaaf67f1e7f>:0
  at System.Threading.Tasks.TaskReplicator+Replica`1[TState].ExecuteAction (System.Boolean& yieldedBeforeC                                                                            ompletion) [0x00000] in <12b418a7818c4ca0893feeaaf67f1e7f>:0
  at System.Threading.Tasks.TaskReplicator+Replica.Execute () [0x00029] in <12b418a7818c4ca0893feeaaf67f1e                                                                            7f>:0
   --- End of inner exception stack trace ---
  at System.Threading.Tasks.TaskReplicator.Run[TState] (System.Threading.Tasks.TaskReplicator+Replicatable                                                                            UserAction`1[TState] action, System.Threading.Tasks.ParallelOptions options, System.Boolean stopOnFirstFai                                                                            lure) [0x0006a] in <12b418a7818c4ca0893feeaaf67f1e7f>:0
  at System.Threading.Tasks.Parallel.PartitionerForEachWorker[TSource,TLocal] (System.Collections.Concurre                                                                            nt.Partitioner`1[TSource] source, System.Threading.Tasks.ParallelOptions parallelOptions, System.Action`1[                                                                            T] simpleBody, System.Action`2[T1,T2] bodyWithState, System.Action`3[T1,T2,T3] bodyWithStateAndIndex, Syst                                                                            em.Func`4[T1,T2,T3,TResult] bodyWithStateAndLocal, System.Func`5[T1,T2,T3,T4,TResult] bodyWithEverything,                                                                             System.Func`1[TResult] localInit, System.Action`1[T] localFinally) [0x001a9] in <12b418a7818c4ca0893feeaaf                                                                            67f1e7f>:0
--- End of stack trace from previous location where exception was thrown ---

  at System.Threading.Tasks.Parallel.ThrowSingleCancellationExceptionOrOtherException (System.Collections.                                                                            ICollection exceptions, System.Threading.CancellationToken cancelToken, System.Exception otherException) [                                                                            0x00010] in <12b418a7818c4ca0893feeaaf67f1e7f>:0
  at System.Threading.Tasks.Parallel.PartitionerForEachWorker[TSource,TLocal] (System.Collections.Concurre                                                                            nt.Partitioner`1[TSource] source, System.Threading.Tasks.ParallelOptions parallelOptions, System.Action`1[                                                                            T] simpleBody, System.Action`2[T1,T2] bodyWithState, System.Action`3[T1,T2,T3] bodyWithStateAndIndex, Syst                                                                            em.Func`4[T1,T2,T3,TResult] bodyWithStateAndLocal, System.Func`5[T1,T2,T3,T4,TResult] bodyWithEverything,                                                                             System.Func`1[TResult] localInit, System.Action`1[T] localFinally) [0x00212] in <12b418a7818c4ca0893feeaaf                                                                            67f1e7f>:0
  at System.Threading.Tasks.Parallel.ForEachWorker[TSource,TLocal] (System.Collections.Generic.IEnumerable                                                                            `1[T] source, System.Threading.Tasks.ParallelOptions parallelOptions, System.Action`1[T] body, System.Acti                                                                            on`2[T1,T2] bodyWithState, System.Action`3[T1,T2,T3] bodyWithStateAndIndex, System.Func`4[T1,T2,T3,TResult                                                                            ] bodyWithStateAndLocal, System.Func`5[T1,T2,T3,T4,TResult] bodyWithEverything, System.Func`1[TResult] loc                                                                            alInit, System.Action`1[T] localFinally) [0x00056] in <12b418a7818c4ca0893feeaaf67f1e7f>:0
  at System.Threading.Tasks.Parallel.ForEach[TSource] (System.Collections.Generic.IEnumerable`1[T] source,                                                                             System.Threading.Tasks.ParallelOptions parallelOptions, System.Action`1[T] body) [0x00033] in <12b418a781                                                                            8c4ca0893feeaaf67f1e7f>:0
  at RawTools.Algorithms.ExtractData.Extract.ScanIndicesPrecursorsDependents (ThermoFisher.CommonCore.Data                                                                            .Interfaces.IRawFileThreadManager rawFileAccessor, System.Int32 MaxProcesses) [0x0010e] in <651c04d8311845                                                                            c091809ac79a3ed047>:0
  at RawTools.WorkFlows.WorkFlowsDDA.UniversalDDA (ThermoFisher.CommonCore.Data.Interfaces.IRawFileThreadM                                                                            anager rawFileThreadManager, RawTools.WorkFlows.WorkflowParameters parameters, RawTools.QC.QcDataCollectio                                                                            n qcDataCollection) [0x00081] in [... cut ...]
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop & RawTools information:**
 - Ubuntu 20
- RawTools version: 2.0.3

**Additional context**
the command:
mono RawTools.exe -d ~/host/calvin/new_data/036_2021_Platinum_DN -qc ~/host/calvin/new_data/036_2021_Platinum_DN/qc_output/ works fine 

link to raw file used
https://dmail-my.sharepoint.com/:i:/g/personal/mtinti_dundee_ac_uk/ET-pleoge49CivASXzgEkJwBqZl4-MutUaLKMt_di-ulgQ?e=E19N19

Thanks for your attention!

kevinkovalchik, 2021-04-01:
That's weird. It looks like an issue with setting the cursor position in the console. Does this happen on every file you try to run or just this one?

mtinti, 2021-04-01:
I just removed that file from the folder and now it's seems working... I'll let you know if it complete the analysis.

mtinti, 2021-04-02:
The program finished ok for all the other files.
Just moving folder for the file that was failing solved the issue, pretty odd indeed.

closed 2021-04-02 by mtinti