kevinkovalchik/RawTools #59: improvement for BoxCar

fazeliniah, 2021-04-16:
Dear developer team, 
Thank you very much for this nice tool. 
I tried using your current RawTools version for a BoxCar raw file but it crashed. I am just wondering if we can improve the QC in RawTools to process the BoxCar scans. 
Thanks

chrishuges, 2021-04-16:
Hi,

I am actually a bit surprised that it didn't work. RawTools is primarily using relationships between MS1 and MS2 scans to calculate many of its metrics, which should be fine here since an MS1 survey is still used to trigger the MS2 scans, and the BoxCar scans are kind of standalone. We have definitely used RawTools internally for BoxCar, but maybe we never pushed these features to the live version. 

What is the error that you get? Also, what version of RawTools are you using? What is the command you use to trigger RawTools when it fails? Can you confirm what type of BoxCar method you are using? MS1 survey > BoxCar range 1 > BoxCar range 2 > MS2 scans?

Thanks,
Chris

fazeliniah, 2021-04-21:
Hi Chris, 
Thanks for getting back to me quickly. 
I am using the latest version of Rawtools and MaxQuant.live to capture the BoxCar data. This is the screenshot of the error. I am also happy to share the BoxCar raw file if you are interested.  
Thanks
[image]

kevinkovalchik, 2021-04-21:
I'm not very familiar with maxquant.live. Is it actually controlling the file creation in some way?

The "orphan scans" up at the top look pretty funky. Does it just go on and on? Like all the MS2 scans are there? When a MS1 precursor scan isn't found for an MS2 scan we exclude the MS2 scan. This seems like a weird thing to happen, but sometimes there are MS2 scans that don't have have precursor scans indicated. This is usually not a problem as it doesn't happen very often. But it looks like maybe the boxcar method here is somehow interfering with this. @chrishuges, what do you think?

chrishuges, 2021-04-21:
It does control the file creation, but I don't think this should really be an issue. It is almost as though it is discarding all of the MS2 scans because it thinks they are orphan so it just fails because it has no precursors to search for. 

As far as I understand BoxCar, it does an MS1 scan from which it selects the MS2 targets. But, before it does dependent scans it does two (or three) rounds of targeted-SIM scans. So, I believe it ends up as MS1 - tSIM - tSIM - MS2. Maybe RawTools is looking in the wrong spot for the precursor because of this modified scan pattern?

I would need to look at the raw file to see what is happening.

fazeliniah, 2021-04-21:
That seems to be correct. The scan pattern is like as MS1-MSX-MSX-MSX-MS2...
You can find a raw file in this link: https://www.dropbox.com/s/1qhedtsisgtrye7/BoxCar_sample.raw?dl=0