kevinkovalchik/RawTools #60: Using Matrix data with quant to assign spectra to TMT-10 plex samples

andreabc, 2021-04-29:
Hello,

Thanks for creating an awesome tool and wiki! I'm new to mass spec analysis and was wondering if it is possible to use the reporter ion information (e.g. intensity/resolution) in the Matrix quant data to assign a spectrum to one or more of the TMT labels? Would it make sense to take the maximum intensity of all the reporter ions?  

Thanks a bunch,
Andrea

chrishuges, 2022-02-02:
Hi Andrea,

I am not sure exactly what you mean here. Can you provide an example of what result you were hoping to get so I can better judge if RawTools can help you?

Thanks,
Chris

chrishuges, 2022-04-21:
Closed due to lack of response. Will re-open if desired.

closed 2022-04-21 by chrishuges