kevinkovalchik/RawTools #61: Chenge mass tolerances for X!tandem search

GeorgTascher, 2021-05-03:
First of all, thanks for this great tool! I use it a lot for QC in our lab, really helpful!
I tried to change the settings for MSMS-tolerance for the QC-runs on our Qexactive in "RawTools_custom/default_config.xml", but RawTools (or X!tandem) always returns to the original settings. I`m using the command line, not the GUI btw.
How can I change the settings permanently?
Cheers!
Georg

chrishuges, 2021-05-03:
Hi,

I believe these are hard-coded and are written on-the-fly based on what RawTools detects as your data type (e.g. Orbitrap MS2 vs Ion Trap MS2). So, you are not able to change these permanently. It was done this way in the QC implementation because we wanted to ensure directories with multiple data types could be tracked in a single QC instance (e.g. a directory containing orbitrap and ion trap MS2 data).

Are the hard coded defaults way off from where you would like them to be?

Cheers,
Chris

GeorgTascher, 2021-05-03:
Hi Chris,

thanks for the quick reply!
This makes sense. But are you sure it is not one single value that is used? RawTools (or X!tandem) also writes a new config.xml in which the MSMS tolerance is set to 0.5 Da, also if Orbitrap MS2-data was used as input. I am not sure if this is actually used in the search, though. A MSMS-mass error plot/metric could be helpful here, but I am not sure if this can be extracted from the X!tandem-results...aaaaaand....feature request! ;-)
The reason why I went into this was the fact that RawTools gives much fewer PSMs compared to other tools like MaxQuant or Proteome Discoverer. But only for Orbitrap-MS2-data (Qexactive), for IT-MS2-data (Lumos), the numbers are comparably high. So this (the fragment mass tolerance) was my first guess.
Regarding your question, the default for IT-data is ok (0.5 Da), but for OT-MS2-data, I would go for 0.02 Da (or 20ppm), but as I said above, I cannot really tell what`s RawTools`s default for OT-MS2-data and if it is different.

Cheers,
Georg

kevinkovalchik, 2021-05-03:
Hi Georg,

There are two parameter files involved in the x!tandem search. There should
be a default one, which contains parameters common to low and high
resolution ms2, and another one that has the parameters specific to the
analysis on hand. At least, that is what I remember. We'll have to make
sure the correct numbers are in there, but I don't recall this ever being
an issue.

When you say the numbers are lower, do you mean the absolute number of IDs
in the QC search? Or is it the the number of targets above a certain decoy
score?

As for the mass accuracy, we are calculating that, but I don't think it is
on a per-scan basis. I think we report median mass error from the QC
search. Does that sound right, Chris?

Kevin

GeorgTascher, 2021-05-03:
Hi Kevin,
Yes, there is the "default_config.xml" and the "custom_config.xml". I checked the custom-file and in there was also the correct name of the .raw which was analyzed, but the Fragment mass tolerance is always set to 0.5 Da (too high for Orbi-MS2 data, as mentioned above). Don`t get me wrong, this is not an issue per se, because I can still track the performance of the instrument. Moreover, I expect more PSMs from MQ and PD because they do much more than X!tandem e.g. mass recalibration using a two-round search. But the large discrepancy for HR-MS2 data seemed a bit odd. That`s why I checked if I can view/change the settings for the search and ran into this...I will double-check when I have time if the difference is not smaller than I think.
And yes, there is a mass accuracy, but it is only in MS1, not in MS2. Also not a big problem.

Also to you, thanks a lot for having developed this tool, Kevin!

Cheers,
Georg

chrishuges, 2021-05-03:
I will have to look into this. I assume you are feeding it some large number of spectra with -N?

chrishuges, 2021-05-03:
Alright I did a QC run on a directory containing two files (one Orbitrap MS2, one Ion Trap MS2, both TMT11 labeled). 

```
mono RawTools.exe -d ./testQC/ -qc ./testQC/qcOutput/ -X ../centOS_tandem/ -db ./uniprot-human_Jan2020.fasta -R -fmods 229.16293@[ -vmods 229.16293@K,15.99491@M -N 1000
2 file(s) to process

Processing: /projects/ptx_analysis/chughes/software/RawTools/rawtools203/./testQC/tmtMs2File.raw

Determing MS analysis order... Done!
Extracting scan indices: 100%
Checking for orphaned scans: 100%

================ Scan indexing report ================

Total scans in file: 65516

Scans linked: 65516

Orphan scans:

None!

All scans accounted for!

======================================================

Extracting reaction events: 100%
Extracting scan data: 100%
Extracting trailer extras: 100%
Extracting precursor masses: 100%
Extracting retention times: 100%
Analyzing precursor peaks: 100%
Refining precursor charge and monoisotopic mass: 100%

Calculating meta data

  MS1 isolation interference

  MS2 scan cycle density

  Ion injection time

  Duty cycle

  Intensity distribution

  Summed intensities

Calculating metrics
Writing MGF file: 100%

Reading FASTA file
Writing target-decoy database: 100%

X! TANDEM Alanine (2017.2.1.4)

Loading spectra (mgf). loaded.
Spectra matching criteria = 979
Starting threads ............................... started.
Computing models:
        testing 1 2 3 testing 1 2 3 testing 1 2 3       sequences modelled = 41 ks
Model refinement:
Merging results:
        from 2345678910111213141516171819202122232425262728293031

Creating report:
        initial calculations  ..... done.
        sorting  ..... done.
        finding repeats ..... done.
        evaluating results ..... done.
        calculating expectations ..... done.
        writing results ..... done.

Valid models = 110

Total hits: 898

Top decoy score: 31.7

Non-decoy hits: 512

Non-decoy hits above top decoy score: 91

Digestion efficiency: 0.747252747252747

Missed cleavage rate (/PSM): 0.318681318681319

IDrate: 0.091

229.16293@K modification frequency: 0.85

15.99491@M modification frequency: 0.161290322580645

Finished writing QC data to csv

QC data written to csv file.

QC file saved successfully

Elapsed time: 220.57 s

Processing: /projects/ptx_analysis/chughes/software/RawTools/rawtools203/./testQC/tmtMs3File.raw

Determing MS analysis order... Done!
Extracting scan indices: 100%
Checking for orphaned scans: 100%

================ Scan indexing report ================

Total scans in file: 31986

Scans linked: 31986

Orphan scans:

None!

All scans accounted for!

======================================================

Extracting reaction events: 100%
Extracting scan data: 100%
Extracting trailer extras: 100%
Extracting precursor masses: 100%
Extracting retention times: 100%
Analyzing precursor peaks: 100%
Refining precursor charge and monoisotopic mass: 100%

Calculating meta data

  MS1 isolation interference

  MS2 scan cycle density

  Ion injection time

  Duty cycle

  Intensity distribution

  Summed intensities

Calculating metrics
Writing MGF file: 100%

X! TANDEM Alanine (2017.2.1.4)

Loading spectra (mgf). loaded.
Spectra matching criteria = 978
Starting threads ............................... started.
Computing models:
        testing 1 2 3 testing 1 2 3 testing 1 2 3       sequences modelled = 41 ks
Model refinement:
Merging results:
        from 2345678910111213141516171819202122232425262728293031

Creating report:
        initial calculations  ..... done.
        sorting  ..... done.
        finding repeats ..... done.
        evaluating results ..... done.
        calculating expectations ..... done.
        writing results ..... done.

Valid models = 366

Total hits: 758

Top decoy score: 21.45

Non-decoy hits: 595

Non-decoy hits above top decoy score: 367

Digestion efficiency: 0.923705722070845

Missed cleavage rate (/PSM): 0.0762942779291553

IDrate: 0.367

229.16293@K modification frequency: 0.893333333333333

15.99491@M modification frequency: 0.0465116279069767

Finished writing QC data to csv

QC data written to csv file.

QC file saved successfully

Elapsed time: 354.64 s

Time to process all 2 files: 00:09:35.2106338
```

I looked in the pep.xml files that are output for each search (at the end of the file you can see the parameters that were used). It looks like it used 0.5 for the orbitrap file and 0.05 for the ion trap file, which is the opposite of what we want. I will try to push a correction for this in the next day or two.

Chris

kevinkovalchik, 2021-05-04:
Seems like we would have noticed that at some point, but apparently not...
I checked the code and this is indeed the case. Check out lines 75-82 in
RawTools/QC/XTandem.cs

Kevin

chrishuges, 2021-05-04:
Yeah, I am a bit surprised we never noticed it. I fixed it in the latest release, [version 2.0.3a](https://github.com/kevinkovalchik/RawTools/releases/tag/2.0.3a).

chrishuges, 2021-05-04:
On the same files:

```
mono RawTools.exe -d ./testQc/ -qc ./testQc/qcOutput/ -X ../centOS_tandem/ -db ./uniprot-human_Jan2020.fasta -R -fmods 229.16293@[ -vmods 229.16293@K,15.99491@M -N 1000
2 file(s) to process

Processing: /projects/ptx_analysis/chughes/software/RawTools/rawtools203a/./testQc/tmtMs2File.raw

Determing MS analysis order... Done!
Extracting scan indices: 100%
Checking for orphaned scans: 100%

================ Scan indexing report ================

Total scans in file: 65516

Scans linked: 65516

Orphan scans:

None!

All scans accounted for!

======================================================

Extracting reaction events: 100%
Extracting scan data: 100%
Extracting trailer extras: 100%
Extracting precursor masses: 100%
Extracting retention times: 100%
Analyzing precursor peaks: 100%
Refining precursor charge and monoisotopic mass: 100%

Calculating meta data

  MS1 isolation interference

  MS2 scan cycle density

  Ion injection time

  Duty cycle

  Intensity distribution

  Summed intensities

Calculating metrics
Writing MGF file: 100%

Reading FASTA file
Writing target-decoy database: 100%

X! TANDEM Alanine (2017.2.1.4)

Loading spectra (mgf). loaded.
Spectra matching criteria = 986
Starting threads ............................... started.
Computing models:
        testing 1 2 3 testing 1 2 3 testing 1 2 3       sequences modelled = 41 ks
Model refinement:
Merging results:
        from 2345678910111213141516171819202122232425262728293031

Creating report:
        initial calculations  ..... done.
        sorting  ..... done.
        finding repeats ..... done.
        evaluating results ..... done.
        calculating expectations ..... done.
        writing results ..... done.

Valid models = 133

Total hits: 851

Top decoy score: 22.65

Non-decoy hits: 514

Non-decoy hits above top decoy score: 147

Digestion efficiency: 0.877551020408163

Missed cleavage rate (/PSM): 0.149659863945578

IDrate: 0.147

229.16293@K modification frequency: 1

15.99491@M modification frequency: 0.170731707317073

Finished writing QC data to csv

QC data written to csv file.

QC file saved successfully

Elapsed time: 208.17 s

Processing: /projects/ptx_analysis/chughes/software/RawTools/rawtools203a/./testQc/tmtMs3File.raw

Determing MS analysis order... Done!
Extracting scan indices: 100%
Checking for orphaned scans: 100%

================ Scan indexing report ================

Total scans in file: 31986

Scans linked: 31986

Orphan scans:

None!

All scans accounted for!

======================================================

Extracting reaction events: 100%
Extracting scan data: 100%
Extracting trailer extras: 100%
Extracting precursor masses: 100%
Extracting retention times: 100%
Analyzing precursor peaks: 100%
Refining precursor charge and monoisotopic mass: 100%

Calculating meta data

  MS1 isolation interference

  MS2 scan cycle density

  Ion injection time

  Duty cycle

  Intensity distribution

  Summed intensities

Calculating metrics
Writing MGF file: 100%

X! TANDEM Alanine (2017.2.1.4)

Loading spectra (mgf). loaded.
Spectra matching criteria = 971
Starting threads ............................... started.
Computing models:
        testing 1 2 3 testing 1 2 3 testing 1 2 3       sequences modelled = 41 ks
Model refinement:
Merging results:
        from 2345678910111213141516171819202122232425262728293031

Creating report:
        initial calculations  ..... done.
        sorting  ..... done.
        finding repeats ..... done.
        evaluating results ..... done.
        calculating expectations ..... done.
        writing results ..... done.

Valid models = 358

Total hits: 849

Top decoy score: 32.9

Non-decoy hits: 637

Non-decoy hits above top decoy score: 335

Digestion efficiency: 0.86865671641791

Missed cleavage rate (/PSM): 0.158208955223881

IDrate: 0.335

229.16293@K modification frequency: 0.806584362139918

15.99491@M modification frequency: 0.218181818181818

Finished writing QC data to csv

QC data written to csv file.

QC file saved successfully

Elapsed time: 255.24 s

Time to process all 2 files: 00:07:43.4085887
```

So, seems to improve the MS2 results a bit. This isn't really an ideal file where I expect it to get a good MS2 ID rate. Perhaps @GeorgTascher you can test this latest version and see if it is working any better for you?

Chris

GeorgTascher, 2021-05-04:
Thanks for looking into it. I will check it out and let you know

GeorgTascher, 2021-05-04:
Hi,
I tried the new 2.0.3a and it is now what I expected. 20-25% gain in PSMs for the QE.
Thank you very much and keep up the good work!

Cheers,
Georg

chrishuges, 2021-05-04:
Great, thanks for checking and for filing this issue in the first place as I doubt we would have noticed it!

Chris

closed 2021-05-04 by chrishuges