kevinkovalchik/RawTools #63: Does RawTools work with FAIMS files

CatFrancoMRC, 2021-05-12:
I am trying to get the metrix on a raw file collected with FAIMS but it keeps failing. Does RawTools support FAIMS?

chrishuges, 2021-05-12:
Hi, 

It should work with FAIMS. Can you please provide the error you are seeing as well as some details about the file where it is failing (e.g. DDA-MS2 3CV FAIMS method) so we can better assist you? Can you also let us know how you are using RawTools (e.g. GUI on Windows, command line on Linux)?

Thanks,
Chris

CatFrancoMRC, 2021-05-13:
Hi, I tried parsing a single FAIMS file and also an RTS one and it works but when I have a go at a more populated location it fails. The one that failed was actually the one that worked perfectly when parsed as a single file inside the folder.

Here is the error

C:\Users\Proteome Discoverer>RawTools -d "Z:\Eclipse\Raw data\RawTools_RTS_test" -qc "Z:\Eclipse\Raw data\RawTools_RTS_test"
28 file(s) to process

Processing: Z:\Eclipse\Raw data\RawTools_RTS_test\CF_hyperplexTESTsample3_labelSwap_FDR_CO_Mix1.raw

Determing MS analysis order... Done!
Extracting scan indices: 99%
Unhandled Exception: System.AggregateException: One or more errors occurred. ---> System.NullReferenceException: Object reference not set to an instance of an object.
   at RawTools.Algorithms.ExtractData.Extract.<>c__DisplayClass0_0.<ScanIndicesPrecursorsDependents>b__1(IEnumerable`1 batch) in C:\Users\chris\OneDrive\Documents\bccrc\softwareWorkspace\RawTools\RawTools\Algorithms\Extraction\ExtractData.cs:line 135
   at System.Threading.Tasks.Parallel.<>c__DisplayClass42_0`2.<PartitionerForEachWorker>b__1()
   at System.Threading.Tasks.Task.InnerInvokeWithArg(Task childTask)
   at System.Threading.Tasks.Task.<>c__DisplayClass176_0.<ExecuteSelfReplicating>b__0(Object <p0>)
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at System.Threading.Tasks.Parallel.PartitionerForEachWorker[TSource,TLocal](Partitioner`1 source, ParallelOptions parallelOptions, Action`1 simpleBody, Action`2 bodyWithState, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithEverything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEachWorker[TSource,TLocal](IEnumerable`1 source, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithEverything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEach[TSource](IEnumerable`1 source, ParallelOptions parallelOptions, Action`1 body)
   at RawTools.Algorithms.ExtractData.Extract.ScanIndicesPrecursorsDependents(IRawFileThreadManager rawFileAccessor, Int32 MaxProcesses) in C:\Users\chris\OneDrive\Documents\bccrc\softwareWorkspace\RawTools\RawTools\Algorithms\Extraction\ExtractData.cs:line 177
   at RawTools.WorkFlows.WorkFlowsDDA.UniversalDDA(IRawFileThreadManager rawFileThreadManager, WorkflowParameters parameters, QcDataCollection qcDataCollection) in C:\Users\chris\OneDrive\Documents\bccrc\softwareWorkspace\RawTools\RawTools\WorkFlows\DDA.cs:line 71
   at RawTools.Program.Run(Dictionary`2 opts) in C:\Users\chris\OneDrive\Documents\bccrc\softwareWorkspace\RawTools\RawTools\Program.cs:line 221
   at RawTools.Program.Main(String[] args) in C:\Users\chris\OneDrive\Documents\bccrc\softwareWorkspace\RawTools\RawTools\Program.cs:line 73

C:\Users\Proteome Discoverer>

soerendip, 2021-07-30:
We don't have observed any issues using FAIMS so far.

chrishuges, 2022-04-26:
Did you ever get this to work? The latest RawTools release offers some updated functionality for FAIMS files. There must be a small bug in the directory processing code. I can try to track this down.