kevinkovalchik/RawTools #64: System.IndexOutOfRangeException: Index was outside the bounds of the array

Nils342, 2021-07-06:
**Describe the bug**
Index out of range exception appears when processing individual .Raw file

**Description of raw file**
In other issues I saw a similar issue which appeared to arise due to the fact that the .raw file was generated through direct infusion using a nanomate. The current file is generated in a similar manner.  

**Command line arguments**
mono RawTools.exe -f /home/nils/Raw_files_for_RawTools/RES_20210630-NM4.raw -x 

**Command line output**
Please paste the entire output from the command line, if applicable (i.e. everything that appeared after you hit enter until things stopped working).
```
total scans in file: 319

Scans linked: 319

Orphan scans:

None!

All scans accounted for!

======================================================

Extracting reaction events: 100%                                                                                                                                                                           
                                                                                                                                                                                                          
Unhandled Exception:
System.IndexOutOfRangeException: Index was outside the bounds of the array.
  at RawTools.Algorithms.ExtractData.Extract.MethodData (ThermoFisher.CommonCore.Data.Interfaces.IRawDataPlus rawFile, RawTools.Data.Containers.ScanIndex index) [0x000a4] in <c730371a78ae4258bcdd6e1d76ea91e3>:0 
  at RawTools.WorkFlows.WorkFlowsDDA.UniversalDDA (ThermoFisher.CommonCore.Data.Interfaces.IRawFileThreadManager rawFileThreadManager, RawTools.WorkFlows.WorkflowParameters parameters, RawTools.QC.QcDataCollection qcDataCollection) [0x000c8] in <c730371a78ae4258bcdd6e1d76ea91e3>:0 
  at RawTools.Program.Run (System.Collections.Generic.Dictionary`2[TKey,TValue] opts) [0x003d3] in <c730371a78ae4258bcdd6e1d76ea91e3>:0 
  at RawTools.Program.Main (System.String[] args) [0x000b3] in <c730371a78ae4258bcdd6e1d76ea91e3>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.IndexOutOfRangeException: Index was outside the bounds of the array.
  at RawTools.Algorithms.ExtractData.Extract.MethodData (ThermoFisher.CommonCore.Data.Interfaces.IRawDataPlus rawFile, RawTools.Data.Containers.ScanIndex index) [0x000a4] in <c730371a78ae4258bcdd6e1d76ea91e3>:0 
  at RawTools.WorkFlows.WorkFlowsDDA.UniversalDDA (ThermoFisher.CommonCore.Data.Interfaces.IRawFileThreadManager rawFileThreadManager, RawTools.WorkFlows.WorkflowParameters parameters, RawTools.QC.QcDataCollection qcDataCollection) [0x000c8] in <c730371a78ae4258bcdd6e1d76ea91e3>:0 
  at RawTools.Program.Run (System.Collections.Generic.Dictionary`2[TKey,TValue] opts) [0x003d3] in <c730371a78ae4258bcdd6e1d76ea91e3>:0 
  at RawTools.Program.Main (System.String[] args) [0x000b3] in <c730371a78ae4258bcdd6e1d76ea91e3>:0 

```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop & RawTools information:**
 - OS: [e.g. iOS]
- RawTools version: [e.g. 1.3.3]

**Additional context**
I tried to upload the .Raw file as a compressed .zip file though for some reason I get the following error:
Is not included in the list.

JonatanFernandez, 2022-01-19:
Hello,

I am also getting a similar error, another out of bounds in the array index. Please find attached the screenshot with the error:

[image]

These are RAW files from a metabolomics experiment but I only wanted the TICs.. (even if I uncheck _Parse matrix_ and _Metrics_ I get the same error)

[image]

Best regards,

chrishuges, 2022-01-19:
Hi,

What is the scan layout in your raw file (e.g. MS and MS/MS scans)? What is
the instrument? You are also using the nanoMate?

Thanks,
Chris

JonatanFernandez, 2022-03-14:
Hi Chris, 

The layout is only MS1, as in many Metabolomics experiments, were you only run MS/MS for a few samples of the whole batch. The instument is a Thermo Q Exactive...  and no nanoMate.  The RAW files should will be available in a public DB in a few months.. cannot share until then; but let me know if I can provide you any other info. 

Best regards,
Jonatan

chrishuges, 2022-03-15:
So there are no MS2 scans in the file that is crashing? If so, that is the issue.

chrishuges, 2022-03-15:
You mentioned you just wanted the TIC. If you select just 'MS1' and 'TIC' in the GUI it still throws the same error (no selection of other options)?

valli28, 2022-04-21:
We have the same issue, with raw files from a Thermo QQQ and an old QE. Some of our files contain only MS1 scans (we are a chromatography company) and RawTools fails to extract basepeak chromatograms, e.g. with the command: `mono RawTools.exe -f FILENAME.raw -chro 1B`. Any workaround/what is the cause for this behaviour?

We either get the
`Unhandled Exception:
System.IndexOutOfRangeException: Index was outside the bounds of the array.` error or the 

`Unhandled Exception:
System.ArgumentOutOfRangeException: Index was out of range. Must be non-negative and less than the size of the collection.` error.

I've only tested this on Linux, Ubuntu 20 so far. 
Any suggestions are welcome :) 

Thanks 
Valli

chrishuges, 2022-04-21:
Hi,

This is because there is only MS1 scans in the file. RawTools is looking
for MS2 by default and doesn’t know what to do when there are none.

We could make an MS1 only mode that just spits out chromatograms, but I
would need a file I could test with if someone is willing to provide one.

Thanks,
Chris

chrishuges, 2022-04-21:
Please try this [pre-release](https://github.com/kevinkovalchik/RawTools/releases/tag/2.0.4a) to see if it fixes your issue. 

You will use 'ms1 only mode' by using the -ms1 flag along with the -chro flag. Right now that is all that is supported. It works fine for me with files that contain MS2 scans, but I don't have any that only contain MS1 scans so I can't test this. The GUI version should also work if you want to test that as well.

chrishuges, 2022-04-21:
I found a random metabolomics file online and it seems it still has some issues with this format. I will keep testing to figure out a solution.

chrishuges, 2022-04-21:
OK I updated the files at the [release link](https://github.com/kevinkovalchik/RawTools/releases/tag/2.0.4a). This one works with files only containing MS1 scans based on my tests. Let me know if you have issues.

valli28, 2022-04-22:
Wow, thanks for being so responsive and proactive @chrishuges !!
We are in vastly different timezones, it would seem, so this correspondence might take a while :D 

I also tested the new version in this release link: 
> OK I updated the files at the [release link](https://github.com/kevinkovalchik/RawTools/releases/tag/2.0.4a). This one works with files only containing MS1 scans based on my tests. Let me know if you have issues.

This version works for our QE files, but when I try the same command for a .raw file from a QQQ, I get one of the same errors I mentioned in my previous comment. 

We are more than happy to provide an example QQQ .raw file for you if you would be interested. 
I've uploaded it to a Google Drive folder here: https://drive.google.com/file/d/1gPJzOhMjsLwBqIi8FNc8gmdr43cLWCKb/view?usp=sharing

I don't know whether I should have made this a seperate issue, as this is a different error message, but I thought it was the same error until now.
I hope it's okay that I just comment for now.

The full error message is as follows:

```
~/workspace/RawTools/RawTools_204a/RawTools-2.0.4a$ mono RawTools.exe -ms1 -f /home/USER/workspace/RawTools/200fm_BSA_100_FS_01_220409050657.raw -chro 1B

Processing: /home/USER/workspace/RawTools/200fm_BSA_100_FS_01_220409050657.raw

Determing MS analysis order... Done!
Extracting scan indices: 100%                                                                                                                    
Checking for orphaned scans: 100%                                                                                                                
                                                                                                                                                 

================ Scan indexing report ================

Total scans in file: 1301

Scans linked: 1301

Orphan scans:

None!

All scans accounted for!

======================================================

Extracting reaction events: 0%
Unhandled Exception:
System.ArgumentOutOfRangeException: Index was out of range. Must be non-negative and less than the size of the collection.
Parameter name: index
  at System.Collections.Generic.List`1[T].get_Item (System.Int32 index) [0x00009] in <533173d24dae460899d2b10975534bb0>:0 
  at ThermoFisher.CommonCore.RawFileReader.StructWrappers.Scan.ScanEvent.ThermoFisher.CommonCore.Data.Interfaces.IScanEventBase.GetReaction (System.Int32 index) [0x00006] in <38096e861d444fd896cbda1ff335437f>:0 
  at RawTools.Algorithms.ExtractData.Extract.ScanEvents (ThermoFisher.CommonCore.Data.Interfaces.IRawDataPlus rawFile, RawTools.Data.Containers.ScanIndex index) [0x0005c] in <4e859c3ff5c04bb7a0c8291f59c0db7a>:0 
  at RawTools.WorkFlows.WorkFlowsSurveyScanOnly.UniversalMs1 (ThermoFisher.CommonCore.Data.Interfaces.IRawFileThreadManager rawFileThreadManager, RawTools.WorkFlows.WorkflowParameters parameters) [0x00098] in <4e859c3ff5c04bb7a0c8291f59c0db7a>:0 
  at RawTools.Program.Run (System.Collections.Generic.Dictionary`2[TKey,TValue] opts) [0x0054c] in <4e859c3ff5c04bb7a0c8291f59c0db7a>:0 
  at RawTools.Program.Main (System.String[] args) [0x000bc] in <4e859c3ff5c04bb7a0c8291f59c0db7a>:0 
[ERROR] FATAL UNHANDLED EXCEPTION: System.ArgumentOutOfRangeException: Index was out of range. Must be non-negative and less than the size of the collection.
Parameter name: index
  at System.Collections.Generic.List`1[T].get_Item (System.Int32 index) [0x00009] in <533173d24dae460899d2b10975534bb0>:0 
  at ThermoFisher.CommonCore.RawFileReader.StructWrappers.Scan.ScanEvent.ThermoFisher.CommonCore.Data.Interfaces.IScanEventBase.GetReaction (System.Int32 index) [0x00006] in <38096e861d444fd896cbda1ff335437f>:0 
  at RawTools.Algorithms.ExtractData.Extract.ScanEvents (ThermoFisher.CommonCore.Data.Interfaces.IRawDataPlus rawFile, RawTools.Data.Containers.ScanIndex index) [0x0005c] in <4e859c3ff5c04bb7a0c8291f59c0db7a>:0 
  at RawTools.WorkFlows.WorkFlowsSurveyScanOnly.UniversalMs1 (ThermoFisher.CommonCore.Data.Interfaces.IRawFileThreadManager rawFileThreadManager, RawTools.WorkFlows.WorkflowParameters parameters) [0x00098] in <4e859c3ff5c04bb7a0c8291f59c0db7a>:0 
  at RawTools.Program.Run (System.Collections.Generic.Dictionary`2[TKey,TValue] opts) [0x0054c] in <4e859c3ff5c04bb7a0c8291f59c0db7a>:0 
  at RawTools.Program.Main (System.String[] args) [0x000bc] in <4e859c3ff5c04bb7a0c8291f59c0db7a>:0
```

chrishuges, 2022-04-22:
Thank you for sharing your file, it was very helpful. The latest version posted [here](https://github.com/kevinkovalchik/RawTools/releases/tag/2.0.4a) works for QQQ files in my hands. Please let me know if you have any issues.

Chris

valli28, 2022-04-25:
Hey @chrishuges 

No problem, you are the one to thank for these quick responses.
I tested your latest version you uploaded. All my QQQ .raw files work so far, with over 300 files tested so far! 

Thank you very much!

Valli

chrishuges, 2022-04-25:
OK great. I will push these changes in the next official release.

Closing this issue for now as it appears to be solved. If any of the other commenters still have issues, please let me know and we can re-open this.

closed 2022-04-25 by chrishuges

he-hai, 2023-02-22:
#81