kevinkovalchik/RawTools #66: test dataset link broken

ll-portes, 2021-12-09:
**Describe the bug**
(1) The link to EBI Pride web portal for the test data set is broken and, if fixed, (2) it seems there is no file called _ch_23Aug2018_HeLa_Std_1.raw_ there in the project PXD008698. 

**Additional context**
Hi guys, 

I'm installing RawTools following the instructions. However, it seems there are two problems when the instructions sent me to the **"download test dataset" step** in order to download the file _ch_23Aug2018_HeLa_Std_1.raw_. (https://github.com/kevinkovalchik/RawTools/wiki/Download-a-test-dataset-for-processing-with-RawTools-for-Linux):

1. **Problem 1** - the link to "visit the EBI Pride ..." gives me the 404 (page not found) error. As of today, that link is https://www.ebi.ac.uk/pride/archive/projects/PXD008698/files
2. removing the "/files" (or searching for PXD008698) allowed me to find all the project files. Specifically, this link works: https://www.ebi.ac.uk/pride/archive/projects/PXD008698
3.  **Problem 2** - However, there is no file with the name _ch_23Aug2018_HeLa_Std_1.raw_. I even searched for any file with _HeLa_ in its name on the project FTP server, without success (http://ftp.ebi.ac.uk/pride-archive/2018/03/PXD008698/). 

Cheers,
Leo.

kevinkovalchik, 2021-12-09:
Hi Leo,

Thanks for bringing this up. It looks like we had a placeholder in there and forgot to update it after the relevant PRIDE repository was made public... I will fix it.

Here is the correct repository: https://www.ebi.ac.uk/pride/archive/projects/PXD011070

Best wishes,
Kevin

kevinkovalchik, 2021-12-09:
Fixed. Let us know if you find other issues or run into problems.

closed 2021-12-09 by kevinkovalchik