kevinkovalchik/RawTools #67: How to run label-free quant? I guess its -u the parameter, but the results are not including right ions

CandyClass, 2021-12-31:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

chrishuges, 2021-12-31:
Can you please provide some more details? It is indeed the -u flag, but what do you mean it is not including the right ions? Can you provide an example?

chrishuges, 2022-04-21:
Closed due to lack of response. Will re-open if desired.

closed 2022-04-21 by chrishuges