kevinkovalchik/RawTools #69: Output MS3 Scans in MGF

IvanSilbern, 2022-01-20:
Dear Kevin,

Would it be possible to adjust MGF-converter so it also reports MS3 scans and not only MS2? For now, it seems that MS3 scans are skipped and not reported in the MGF file.

Thank you very much for your efforts!

Best regards,
Ivan

kevinkovalchik, 2022-01-20:
What do you think, @chrishuges? It seems it would be pretty easy to add an argument specifying which MS levels get put in the MGF. This would be small enough that I might be able to find time for it.

Kevin

chrishuges, 2022-01-20:
Yes, would be quite straightforward to implement I think. If you have time that would be fantastic. Otherwise I can circle back around to it in a couple of weeks after some grant deadlines.

chrishuges, 2022-02-02:
Please try out [2.0.4](https://github.com/kevinkovalchik/RawTools/releases) to see if this fixes your issue. I opted to not combine them into a single MGF, as they are just text files and could be manually integrated downstream quite easily. But, this functionality could be added in the future if desired.

IvanSilbern, 2022-02-03:
Dear Christopher, dear Kevin,

thank you very much for the prompt update! I tried to convert a few raw files and it worked. Also comparison with mgf produced by msconvert did not show significant differences.
Just a minor note: The "original" rawtool mgf contains more information about the scan, such as peptide mass, charge, and retention time. The new ms2/ms3 mgf files contain only the scan number, it might be also useful to include the mass, charge of the precursor ion, and the retention time, if possible. Thank you very much again for your work!

Best,
Ivan

chrishuges, 2022-02-03:
Thanks Ivan, glad it worked for you. I ended up having to remove the release temporarily, but when I put it back up I will add that other info you requested. I had removed it to streamline the format with MS1 (no mass, charge information) while I was debugging and must have forgotten to put it back in for MS2 and MS3 scans. Thanks for letting me know. I should have the release reposted by the end of the week.

chrishuges, 2022-02-08:
Hi Ivan,

I just re-uploaded 2.0.4 to the releases section and it adds more detail to the MGF output for MS1/MS2/MS3.

Thanks,
Chris

closed 2022-02-09 by chrishuges