kevinkovalchik/RawTools #71: How to run the search with xtandem on linux?

sorenwacker, 2022-02-01:
Hi,

I want to experiment with the search function of RawTools. I tried this command on linux, but it seems no search is started. There is also no error message and I did not find anything about this in the documentation. 

    rawtools.sh -f /data/SA001-R3-A-2ul+f_20191025194801.raw -X /home/.../miniconda3/envs/lrg/bin/ -db /data/Proteomics/testing/QC-testing.faa -r TMT11

I installed xtandem with `miniconda` on Linux. The executable is called just `xtandem` not `xtandem.exe`.

Maybe it would be better to specify the executable rather than the directory in that case?

Are there parameters missing?

chrishuges, 2022-02-01:
Hi,

I guess it is looking for tandem.exe. In the meantime, can you try just renaming the tandem bin file to tandem.exe and see if it works? 

Thanks,
Chris

chrishuges, 2022-04-21:
Assume this issue is fixed due to the lack of response. Will re-open if not.

closed 2022-04-21 by chrishuges