kevinkovalchik/RawTools #77: Using CLParser

biobibibi, 2022-08-28:
Hi 

Thank you for your great work!
I am new to C# and actually trying my best to learn from your excellent code. 
But I could not find the nupkg for the CLParser nor the CLParserForNetFramework in the package folder or the Internet, which was needed in the main program. 
Can you show me where to get that package? 
Thank you so much.

Siwei

kevinkovalchik, 2022-08-29:
Hi Siwei,

Thanks for your interest. Sorry about that missing package. It was something that was developed internally and I never got around to putting it up on github. To clarify, are you looking for the source code, or is the dll actuality missing in a release?

I think Chris might have the source code for it... @chrishuges, do you have the code for the CL parser? It could probably just be placed in a subdirectory of the RawTools repository, both for our reference and for anyone who wants to build the project from source. If you don't have it I'll dig through my archives sometime. 

Best, 
Kevin

biobibibi, 2022-08-29:
Hi Kevin

Thanks you!
I can run the newest 2.0.7 version of RawTools without problem and I am looking for the source code so that I could go through the process of building the project.
I appreciate you and Chris's help.

Best regards.
Siwei

biobibibi, 2022-08-31:
Hi Kevin

I think I just found it under the local RawTools folder.
Thanks again.

Siwei

closed 2022-08-31 by biobibibi

kevinkovalchik, 2022-08-31:
Oh, good. I didn't realize it was in there already.

Best,
Kevin