kevinkovalchik/RawTools #78: Thermo Tribrid data are not parsed

mmross0321, 2022-10-17:
**Describe the bug**
Briefly describe what you were trying to do and what went wrong. Be general, there is room for specifics below.

**Description of raw file**
Provide details about the raw file you are trying to process. e.g. Is it DDA? MS2 or SPS-MS3? How long was the run and how big is the file?

**Command line arguments**
If this is something that happened while running RawTools, enter the exact arguments used on the command line (e.g. RawTools parse -f C:\myfile.raw -mpx -R)

**Command line output**
Please paste the entire output from the command line, if applicable (i.e. everything that appeared after you hit enter until things stopped working).
```
[Paste output here, between the triple quotes]
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop & RawTools information:**
 - OS: [e.g. iOS]
- RawTools version: [e.g. 1.3.3]

**Additional context**
Add any other context about the problem here.

chrishuges, 2022-10-17:
Could you please describe the issue you are encountering and provide information related to the questions posed above?

mmross0321, 2022-10-17:
I tried to parse metrics from Thermo Tribrid data, the process seemed to proceed in Windows command prompt view, but no files were written. I used the GUI.

chrishuges, 2022-10-17:
Hi,

In order to help you I need some more information so I can try to re-create the bug and hopefully solve it. Can you please answer these questions:

Provide details about the raw file you are trying to process (e.g. Is it DDA? MS2 or SPS-MS3)?

Provide details of what commands you have input into the GUI (e.g. which options did you select, which did you not? screenshot here might help).

Does it appear like the command finishes? If there are no files written, it is likely hitting an error. I know in the GUI this can be hard to catch, but there should be a 'rawtools_log.txt' file that gets written into the RawTools directory. Can you please provide me with what is in this file?

Thank you,
Chris

mmross0321, 2022-10-18:
Chris,
Thank you for your reply.
The raw file was acquired using a SPS-MS3 method.
Below first is the GUI parameters I used.  Yesterday, the process seems to run to completion as viewed in the Windows Command Prompt window, but no files were written to the specified folder.

Below second is the output from running the program from the Command Prompt.
This process yielded the attached text file.
Mark

***@***.***

Output from using the Command Prompt:

C:\RawTools-2.0.7\RawTools-2.0.7>rawtools parse -f JM5072_HCT_Cys_TMT4-1.raw -x

Processing: C:\RawTools-2.0.7\RawTools-2.0.7\JM5072_HCT_Cys_TMT4-1.raw

Determing MS analysis order... Done!
Extracting scan indices: 99%
Unhandled Exception: System.AggregateException: One or more errors occurred. ---> System.NullReferenceException: Object reference not set to an instance of an object.
   at RawTools.Algorithms.ExtractData.Extract.<>c__DisplayClass0_0.<ScanIndicesPrecursorsDependents>b__1(IEnumerable`1 batch) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Algorithms\Extraction\ExtractData.cs:line 135
   at System.Threading.Tasks.Parallel.<>c__DisplayClass42_0`2.<PartitionerForEachWorker>b__1()
   at System.Threading.Tasks.Task.InnerInvokeWithArg(Task childTask)
   at System.Threading.Tasks.Task.<>c__DisplayClass176_0.<ExecuteSelfReplicating>b__0(Object <p0>)
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at System.Threading.Tasks.Parallel.PartitionerForEachWorker[TSource,TLocal](Partitioner`1 source, ParallelOptions parallelOptions, Action`1 simpleBody, Action`2 bodyWithState, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithEverything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEachWorker[TSource,TLocal](IEnumerable`1 source, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithEverything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEach[TSource](IEnumerable`1 source, ParallelOptions parallelOptions, Action`1 body)
   at RawTools.Algorithms.ExtractData.Extract.ScanIndicesPrecursorsDependents(IRawFileThreadManager rawFileAccessor, Int32 MaxProcesses) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Algorithms\Extraction\ExtractData.cs:line 177
   at RawTools.WorkFlows.WorkFlowsDDA.UniversalDDA(IRawFileThreadManager rawFileThreadManager, WorkflowParameters parameters, QcDataCollection qcDataCollection) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\WorkFlows\DDA.cs:line 71
   at RawTools.Program.Run(Dictionary`2 opts) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Program.cs:line 243
   at RawTools.Program.Main(String[] args) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Program.cs:line 73

2022-05-04 19:55:02.227 -07:00 [INF] Program started with arguments: -f C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\raw-test-file\200fm_BSA_100_FS_01_220409050657.raw -xic 500,10
2022-05-04 19:55:02.642 -07:00 [INF] Files to be processed, provided as list: C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\raw-test-file\200fm_BSA_100_FS_01_220409050657.raw
2022-05-04 19:55:02.670 -07:00 [INF] Extracting scan indices
2022-05-04 19:55:02.845 -07:00 [INF] Extracting method/instrument information
2022-05-04 19:55:03.086 -07:00 [INF] Extracting trailer extras
2022-05-04 19:55:03.103 -07:00 [INF] Extracting precursor masses
2022-05-04 19:55:03.113 -07:00 [INF] Extracting retention times
2022-05-04 20:00:45.293 -07:00 [INF] Program started with arguments: -f C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\raw-test-file\200fm_BSA_100_FS_01_220409050657.raw -xic 500,10
2022-05-04 20:00:45.405 -07:00 [INF] Files to be processed, provided as list: C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\raw-test-file\200fm_BSA_100_FS_01_220409050657.raw
2022-05-04 20:00:45.433 -07:00 [INF] Extracting scan indices
2022-05-04 20:00:45.605 -07:00 [INF] Extracting method/instrument information
2022-05-04 20:00:45.838 -07:00 [INF] Extracting trailer extras
2022-05-04 20:00:45.852 -07:00 [INF] Extracting precursor masses
2022-05-04 20:00:45.862 -07:00 [INF] Extracting retention times
2022-05-05 13:03:56.704 -07:00 [INF] Program started with arguments: -f C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\raw-test-file\200fm_BSA_100_FS_01_220409050657.raw - -faimsMgf
2022-05-05 13:03:57.071 -07:00 [INF] Files to be processed, provided as list: C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\raw-test-file\200fm_BSA_100_FS_01_220409050657.raw
2022-05-05 13:03:57.098 -07:00 [INF] Extracting scan indices
2022-05-05 13:03:57.269 -07:00 [INF] Extracting method/instrument information
2022-05-05 13:03:57.509 -07:00 [INF] Extracting trailer extras
2022-05-05 13:03:57.523 -07:00 [INF] Extracting precursor masses
2022-05-05 13:03:57.532 -07:00 [INF] Extracting retention times
2022-10-14 09:51:30.527 -04:00 [INF] Program started with arguments: 
[... cut ...]
2022-10-18 10:36:26.578 -04:00 [INF] Files to be processed, provided as list: C:\RawTools-2.0.7\RawTools-2.0.7\JM5072_HCT_Cys_TMT4-1.raw
2022-10-18 10:36:26.621 -04:00 [INF] Extracting scan indices
2022-10-18 10:44:53.677 -04:00 [INF] Program started with arguments: -f C:\RawTools-2.0.7\RawTools-2.0.7\JM5072_HCT_Cys_TMT4-1.raw -x
2022-10-18 10:44:53.812 -04:00 [INF] Files to be processed, provided as list: C:\RawTools-2.0.7\RawTools-2.0.7\JM5072_HCT_Cys_TMT4-1.raw
2022-10-18 10:44:53.856 -04:00 [INF] Extracting scan indices

mmross0321, 2022-10-18:
I did not finish the message…today using the GUI with the parameters shown below resulted in the program stopping (crashing) at the first step, extracting indices.

chrishuges, 2022-10-18:
So, it looks like QC works but maybe parse doesn't? If you run this one file that is giving an error in the GUI, it will fail. If you run another file, also MS3, with the same parameters does it succeed?

I think this is an issue related to orphan scans that I actually recently pushed a solution for (https://github.com/kevinkovalchik/RawTools/tree/branchVersion204), but I haven't incorporated it into a release yet. If you go into your raw file in XCalibur or something, is the last scan in the file an MS2 scan that doesn't have a matched MS3?

chrishuges, 2022-10-18:
That is good, so it is an issue with that one specific file then. The MS2 scans being zero is a bug I know about and will be fixed in the next release, it does not impact the actual processing. 

Just to clarify, the last scan of the raw file that is failing is an MS1 scan?

mmross0321, 2022-10-20:
A raw file that failed during RawTools parsing has an MS2 spectrum as the last scan.
Mark