kevinkovalchik/RawTools #81: Provided test file gives System.ArgumentOutOfRangeException using Windows

VittorioRainaldi, 2023-02-17:
Running `.\RawTools.exe -f C:\RawTools\ch_23Aug2018_HeLa_Std_1.raw -x` on Windows using the latest release gives the following output:
```
Unhandled Exception: System.AggregateException: One or more errors occurred. ---> System.ArgumentOutOfRangeException: The value must be greater than or equal to zero and less than the console's buffer size in that dimension.
Parameter name: top
Actual value was -1.
   at System.Console.SetCursorPosition(Int32 left, Int32 top)
   at RawTools.Utilities.ConsoleUtils.ClearLastLine() in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Utilities\Utilities.cs:line 75
   at RawTools.Utilities.ProgressIndicator.Update() in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Utilities\Utilities.cs:line 199
   at RawTools.Algorithms.ExtractData.Extract.<>c__DisplayClass0_0.<ScanIndicesPrecursorsDependents>b__1(IEnumerable`1 batch) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Algorithms\Extraction\ExtractData.cs:line 172
   at System.Threading.Tasks.Parallel.<>c__DisplayClass42_0`2.<PartitionerForEachWorker>b__1()
   at System.Threading.Tasks.Task.InnerInvokeWithArg(Task childTask)
   at System.Threading.Tasks.Task.<>c__DisplayClass176_0.<ExecuteSelfReplicating>b__0(Object <p0>)
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at System.Threading.Tasks.Parallel.PartitionerForEachWorker[TSource,TLocal](Partitioner`1 source, ParallelOptions parallelOptions, Action`1 simpleBody, Action`2 bodyWithState, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithEverything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEachWorker[TSource,TLocal](IEnumerable`1 source, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithEverything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEach[TSource](IEnumerable`1 source, ParallelOptions parallelOptions, Action`1 body)
   at RawTools.Algorithms.ExtractData.Extract.ScanIndicesPrecursorsDependents(IRawFileThreadManager rawFileAccessor, Int32 MaxProcesses) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Algorithms\Extraction\ExtractData.cs:line 177
   at RawTools.WorkFlows.WorkFlowsDDA.UniversalDDA(IRawFileThreadManager rawFileThreadManager, WorkflowParameters parameters, QcDataCollection qcDataCollection) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\WorkFlows\DDA.cs:line 71
   at RawTools.Program.Run(Dictionary`2 opts) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Program.cs:line 243
   at RawTools.Program.Main(String[] args) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Program.cs:line 73
```
Note that the same command runs without issues on Linux via mono.

he-hai, 2023-02-22:
I got the same error when using VS code on windows. But using power shell directly has no problem. It seems a problem from VS code, see https://github.com/PowerShell/PSReadLine/issues/3216

VittorioRainaldi, 2023-02-24:
I never used VS code, the error is there when using power shell for me

chrishuges, 2023-02-24:
What about if you use command prompt instead of power shell?

VittorioRainaldi, 2023-03-06:
I tried the command prompt and I get the same error. I tried piping the output to a text file and this is what I get:
```
Processing: M:\labeling\RawTools\..\ch_23Aug2018_HeLa_Std_1.raw

Determing MS analysis order... Done!
```