kevinkovalchik/RawTools #86: Error Message with RawTools-2.0.7 on Windows 11

johnmeinert, 2025-05-15:
Hi, I'm trying to run RawTools-2.0.7 on Windows 11 and I followed all the instructions for downloading and processing files, however, when I attempt to process files I get this error message. I've seen threads with similar errors and tried the things suggested there but it hasn't fixed it and I continue to receive the same message. Could I get some help? Thanks!

Unhandled Exception: System.AggregateException: One or more errors occurred. ---> System.ArgumentOutOfRangeException: The value must be greater than or equal to zero and less than the console's buffer size in that dimension.
Parameter name: top
Actual value was -1.
   at System.Console.SetCursorPosition(Int32 left, Int32 top)
   at RawTools.Utilities.ConsoleUtils.ClearLastLine() in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Utilities\Utilities.cs:line 75
   at RawTools.Utilities.ProgressIndicator.Update() in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Utilities\Utilities.cs:line 199
   at RawTools.Algorithms.ExtractData.Extract.<>c__DisplayClass0_0.<ScanIndicesPrecursorsDependents>b__1(IEnumerable`1 batch) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Algorithms\Extraction\ExtractData.cs:line 172
   at System.Threading.Tasks.Parallel.<>c__DisplayClass42_0`2.<PartitionerForEachWorker>b__1()
   at System.Threading.Tasks.Task.InnerInvokeWithArg(Task childTask)
   at System.Threading.Tasks.Task.<>c__DisplayClass176_0.<ExecuteSelfReplicating>b__0(Object <p0>)
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task.ThrowIfExceptional(Boolean includeTaskCanceledExceptions)
   at System.Threading.Tasks.Task.Wait(Int32 millisecondsTimeout, CancellationToken cancellationToken)
   at System.Threading.Tasks.Parallel.PartitionerForEachWorker[TSource,TLocal](Partitioner`1 source, ParallelOptions parallelOptions, Action`1 simpleBody, Action`2 bodyWithState, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithEverything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEachWorker[TSource,TLocal](IEnumerable`1 source, ParallelOptions parallelOptions, Action`1 body, Action`2 bodyWithState, Action`3 bodyWithStateAndIndex, Func`4 bodyWithStateAndLocal, Func`5 bodyWithEverything, Func`1 localInit, Action`1 localFinally)
   at System.Threading.Tasks.Parallel.ForEach[TSource](IEnumerable`1 source, ParallelOptions parallelOptions, Action`1 body)
   at RawTools.Algorithms.ExtractData.Extract.ScanIndicesPrecursorsDependents(IRawFileThreadManager rawFileAccessor, Int32 MaxProcesses) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Algorithms\Extraction\ExtractData.cs:line 177
   at RawTools.WorkFlows.WorkFlowsDDA.UniversalDDA(IRawFileThreadManager rawFileThreadManager, WorkflowParameters parameters, QcDataCollection qcDataCollection) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\WorkFlows\DDA.cs:line 71
   at RawTools.Program.Run(Dictionary`2 opts) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Program.cs:line 243
   at RawTools.Program.Main(String[] args) in C:\Users\chughes\Documents\bccrc\softwareRepository\RawTools\rawtoolsDev\RawTools\Program.cs:line 73

lyons89, 2026-09-11:
I just received the same error on 9/11/2026
Did @johnmeinert ever figure it out?

shimin-chen, 2026-09-14:
I am also seeing a similar error on Windows 11. I ran the same command in a Windows 10 system and was able to get the expected output.

kevinkovalchik, 2026-09-15:
Thanks for reviving the thread. I might have some time next week to look at the problem. Out of curiosity, on windows 11 can you get it to run successfully by using a different console app? Windows switched to their new “Terminal” app sometime in the last couple years, and I wonder if the PowerShell or Command Prompt apps still work or if it is system-wide problem. 

Best,
Kevin

shimin-chen, 2026-09-16:
Hi Kevin, 

Thanks for the continued support on this tool! I have tried the same command on the "Terminal" app (which seems to open PowerShell), Windows PowerShell, and Command Prompt on Windows 11. I received the same error across all three apps (see attached). The same command works on Windows 10 Command Prompt. The RawTools version is 2.0.7.

Best,
Shimin

[Unhandled Exception System.Aggregat.txt](https://github.com/user-attachments/files/32293053/Unhandled.Exception.System.Aggregat.txt)